pothosware/SoapyRemote #14: Raspberry Pi SoapyClientHandler::handleOnce(306) unknown call?

cjcliffe, 2015-10-30:
Not sure what's up yet here; pulled and rebuilt SoapySDR, SoapyRemote and SoapyRTLSDR on master branches at least twice on Pi and OSX plus CubicSDR as well -- Might just be something simple I've overlooked though.

CubicSDR log shows:
```
Enumerating remote address: raspberrypi.local
  available = Yes
  driver = remote
  label = Generic RTL2832U OEM :: 00000003
  manufacturer = Realtek
  product = RTL2838UHIDIR
  remote = tcp://raspberrypi.local:55132
  remote:driver = rtlsdr
  rtl = 0
  serial = 00000003
  tuner = Rafael Micro R820T
Make device 0
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=0
Error making device: RemoteError: SoapyClientHandler::handleOnce(306) unknown call
Reporting enumeration complete.
SDR enumerator done.
[1m[31m[ERROR] ~SoapyRPCUnpacker: Unconsumed payload bytes 7[0m
```

Pi side shows:
```
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

upid://raspberrypi?pid=3036&hid=8323329
Launching the server... tcp://0.0.0.0:55132
Server bound to 0.0.0.0:55132
Press Ctrl+C to stop the server
SoapyServerListener::accept(192.168.1.102:51011)
SoapyServerListener::accept(192.168.1.102:51012)
Found Rafael Micro R820T tuner
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(192.168.1.102:51013)
SoapyServerListener::accept(192.168.1.102:51014)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(192.168.1.102:51015)
SoapyServerListener::accept(192.168.1.102:51016)
Found Rafael Micro R820T tuner
SoapyServerListener::close()
SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv(header) FAIL: No data available
SoapyServerListener::close()
```

Thanks!

guruofquality, 2015-10-30:
> Error making device: RemoteError: SoapyClientHandler::handleOnce(306) unknown call

That would happen if the server was out of date and did not support SOAPY_REMOTE_GET_STREAM_ARGS_INFO      = 306,

cjcliffe, 2015-10-30:
Aye, cleaned my local repo and it just pulled a bunch of stuff this time; user error! :)

guruofquality, 2015-10-30:
I meant to have a compatibility check, it would make failure more graceful, but I forgot about updating it.

cjcliffe, 2015-10-30:
All looks good! :+1: 

Status: Closed/Wontfix  Reason: ID10T error

closed 2015-10-30 by cjcliffe