pothosware/SoapyRemote #26: SoapySDRServer segfault

Geremia, 2017-01-05:
Using the Osmocom source block with Soapy Remote, I sometimes get a segfault:

Here is what GNUradio says:
```
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.8.1
built-in source types: file fcd rtl_tcp sdrplay rfspace soapy redpitaya 
[INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CF32, scaleFactor=32767, mtu=1500, window=44040192)
[INFO] Client side stream bound to 192.168.1.175:47903
[INFO] Client side status bound to 192.168.1.175:45892
[INFO] Using format CS16.
[INFO] Server side stream bound to [::ffff:192.168.1.7]:48810
[INFO] Server side stream connected to [::ffff:192.168.1.175]:47903
[INFO] Server side status connected to [::ffff:192.168.1.175]:45892
[INFO] Configured sender endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=43008 KiB
[INFO] Client side stream connected to 192.168.1.7:48810
[INFO] Configured receiver endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=43008 KiB
[1m[33m[WARNING] Set thread priority 0.5 failed: Operation not permitted[0m
Using Volk machine: avx_64_mmx_orc
[1m[31m[ERROR] SoapyLogAcceptor::handlerLoop() [0m
thread[thread-per-block[0]: <block soapy_source_c (2)>]: SoapyRPCUnpacker::recv(header) FAIL: 
```
Here is what the remote server running SoapySDRServer says right before it segfaults:
```
mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=2500 rev=0206 serno=B0001P0004 bus=001 port=004 devAvail=1
mir_sdr_usb_SetDeviceIdx idx=0 numDevices=1
Opened device with idVendor = 0x1df7 idProduct = 0x2500 fwVersion = 0x0206 busNum = 001 portNum = 004
mir_sdr_DCoffsetIQimbalanceControl: DC:1 IQ:1
mir_sdr_AgcControl: 1 -30 0 0 0 0 1
mir_sdr_StreamInit()
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=20dB fs=7.000MHz rf=479.000MHz bw=6.000MHz if=0.000MHz
DownConvert: Enable=0 DecM=1 OutScale=0 (fs=7.000000 bw=6000 if=0)
mir_sdr_usb_USB DLL: Revision 0.1.1
mir_sdr_2500_Init: fnaddr = 2 detected, trying to change...
mir_sdr_2500_Init: fnaddr = 6
mir_sdr_2500_Init: adjusting squelch trim 0x1, rx gating enable 1, tx_trim 0, reg2 = 0x4801
initHw: Register7 = 0x000085
initRfFreqDependentHw(1): Tuner Register0 = 0x057480
Error: libusb_submit_transfer() -1
mir_sdr_2500_StreamInit: Open failed
initHw: mir_sdr_2500_StreamInit() Error 0x00000001
mir_sdr_Init: initHw returns error 7
```

guruofquality, 2017-02-16:
I must have missed this bug. It looks like the SDR Play disconnected or crashed for some reason. I dont think there is anything to fix on the Soapy side of things

corq, 2017-03-07:
Getting this issue with fresh pull, have pulled/built current versions of SoapySDR SoapySDRPlay and SoapyRemote, no build errors encountered. Soapy remote loads fine, then attempt to connect with CubicSDR remotely from Local LAN:

`/opt/SoapySDR/build# SoapySDRServer --bind="0.0.0.0:1234" 
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

SoapyServerListener::accept(192.168.1.8:46996)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(192.168.1.8:46998)
mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=2500 rev=0206 serno=B0001P0001 bus=001 port=001 devAvail=1
SoapyServerListener::accept(192.168.1.8:47000)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(192.168.1.8:47002)
mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=2500 rev=0206 serno=B0001P0001 bus=001 port=001 devAvail=1
mir_sdr_usb_SetDeviceIdx idx=0 numDevices=1
Opened device with idVendor = 0x1df7 idProduct = 0x2500 fwVersion = 0x0206 busNum = 001 portNum = 001
mir_sdr_usb_ReleaseDeviceIdx 0
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(192.168.1.8:47010)
SoapyServerListener::accept(192.168.1.8:47012)
mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=2500 rev=0206 serno=B0001P0001 bus=001 port=001 devAvail=1
SoapyServerListener::accept(192.168.1.8:47014)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(192.168.1.8:47016)
mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=2500 rev=0206 serno=B0001P0001 bus=001 port=001 devAvail=1
mir_sdr_usb_SetDeviceIdx idx=0 numDevices=1
Opened device with idVendor = 0x1df7 idProduct = 0x2500 fwVersion = 0x0206 busNum = 001 portNum = 001
mir_sdr_StreamInit()
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=40dB fs=2.000MHz rf=6.754MHz bw=1.536MHz if=0.000MHz
DownConvert: Enable=0 DecM=1 OutScale=0 (fs=2.000000 bw=1536 if=0)
mir_sdr_usb_USB DLL: Revision 0.1.1
mir_sdr_2500_Init: fnaddr = 3
mir_sdr_2500_Init: adjusting squelch trim 0x1, rx gating enable 1, tx_trim 0, reg2 = 0x4801
initHw: Register7 = 0x000014
initRfFreqDependentHw(1): Tuner Register0 = 0x04fe10
mir_sdr_SetFs: Sample Freq requested 2000000.000000
mir_sdr_SetFs: Fs->FsNomHz+dFsHz=2000000.0+0.0Hz=2000000.0Hz FsToggle->1
mir_sdr_SetRf: f->126754000.000Hz (int=15 frac=179 afc=0) fSynth:2028064000.000
mir_sdr_SetRf: Rf->RfNomHz+dRfHz+LifHz+Lif1Hz=6754000.0+0.0+0.0Hz+120000000.0Hz=126754000.0Hz RfToggle->1
mir_sdr_SetGr: GR->40[16,0,0,24] gRset->0xD0 DCCALmode=3 DCCALspd=0 GrToggle->1
setToggleStates: initialising sampNum=0xa090031c, gainSetting=0x0d0, FsToggle=0, RfToggle=0, GrToggle=0
setToggleStates: initialising Fs=2000000.000, Rf=126754000.000, Gr=40
mir_sdr_RSP_SetGr: in: 40 1 1 0
mir_sdr_RSP_SetGr: GR->40[40,0,0,24] gRset->0xE8 DCCALmode=3 DCCALspd=0 GrToggle->1
mir_sdr_RSP_SetGr: out: 40 64 1
mir_sdr_DecimateControl: 0 1 0
mir_sdr_DecimateControl: DecimateEnable=0, DecimateDecM=1, DecimateOutScale=0, DecimateRound[0]=0x00000000, DecimateRound[1]=0x00000000
mir_sdr_SetDcMode: DCCAL: mode->4 speedup->0
mir_sdr_SetDcTrackTime: DCTRK_TIM->63
DCcoeff=0.001500 adj_for_Fs=0.001500 Fs=2000000.000
AGCcoeff=0.290000 adj_for_Fs=0.290000 Fs=2000000.000
adj_for_PktSz=0.290000 PktSz=252
mir_sdr_DCoffsetIQimbalanceControl: DC:1 IQ:1
mir_sdr_AgcControl: 1 -30 0 0 0 0 1
mir_sdr_AgcControl: 1 -30 0 0 0 0 1
mir_sdr_DCoffsetIQimbalanceControl: DC:1 IQ:1
mir_sdr_AgcControl: 1 -30 0 0 0 0 1
mir_sdr_AgcControl: 1 -30 0 0 0 0 1
AGCcoeff=0.290000 adj_for_Fs=0.290000 Fs=2000000.000
adj_for_PktSz=0.290000 PktSz=252
mir_sdr_RSP_SetGr: ERROR: previous gain update not received (or timed out)
mir_sdr_RSP_SetGr: ERROR: previous gain update not received (or timed out)
mir_sdr_RSP_SetGr: ERROR: previous gain update not received (or timed out)
mir_sdr_RSP_SetGr: ERROR: previous gain update not received (or timed out)
[... 16 lines cut ...]
mir_sdr_RSP_SetGr: out: 48 72 1
DCoffsetCorrection: switched to tracking mode diff=74 curr=-279.24 last=-353.91
mir_sdr_ReadPacket: Gain update confirmed: Gr=48dB GrToggle=0 gset=0xf0
grChanged @ 1380
mir_sdr_RSP_SetGr: in: -4 1 0 0
mir_sdr_RSP_SetGr: GR->44[44,0,0,24] gRset->0xEC DCCALmode=4 DCCALspd=0 GrToggle->1
mir_sdr_RSP_SetGr: out: 44 68 1
DCoffsetCorrection: switched to tracking mode diff=75 curr=-602.34 last=-526.90
mir_sdr_ReadPacket: Gain update confirmed: Gr=44dB GrToggle=1 gset=0xec
grChanged @ 1749
mir_sdr_RSP_SetGr: in: -4 1 0 0
mir_sdr_RSP_SetGr: GR->40[40,0,0,24] gRset->0xE8 DCCALmode=4 DCCALspd=0 GrToggle->0
mir_sdr_RSP_SetGr: out: 40 64 1
DCoffsetCorrection: switched to tracking mode diff=9 curr=-1327.44 last=-1318.17
mir_sdr_ReadPacket: Gain update confirmed: Gr=40dB GrToggle=0 gset=0xe8
grChanged @ 2125
mir_sdr_RSP_SetGr: in: -3 1 0 0
mir_sdr_RSP_SetGr: GR->37[37,0,0,24] gRset->0xE5 DCCALmode=4 DCCALspd=0 GrToggle->1
mir_sdr_RSP_SetGr: out: 37 61 1
DCoffsetCorrection: switched to tracking mode diff=0 curr=-2105.41 last=-2105.95
mir_sdr_ReadPacket: Gain update confirmed: Gr=37dB GrToggle=1 gset=0xe5
grChanged @ 2503
DCoffsetCorrection: switched to tracking mode diff=55 curr=-2403.00 last=-2347.42
Segmentation fault
`

guruofquality, 2017-03-07:
Hmm... Can you run SoapySDRServer in gdb? Possibly build with -DCMAKE_BUILD_TYPE=Debug as well. I think a traceback here will be helpful. Basically we need to know where is crashing and that the input may have been so we can narrow it down.

AnsgarSchmidt, 2018-01-16:
Hi,
same problem here, I used the latest git master for SoapySDR, SoapySDRPlay and SoapyRemote.

Running on raspberry pi 3 with SDRplay as receiver and cubicSDR on Mac. 
The gdb outbut is:

Starting program: /usr/local/bin/SoapySDRServer --bind
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

c4288117-626f-150a-8567-19aa007f0101
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server...
[New Thread 0x76bc2050 (LWP 6573)]
[New Thread 0x763c2050 (LWP 6574)]
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]
Press Ctrl+C to stop the server
SoapyServerListener::accept([::ffff:192.168.1.86]:52398)
[New Thread 0x759ff050 (LWP 6575)]
SoapyServerListener::accept([::ffff:192.168.1.86]:52399)
[New Thread 0x74fff050 (LWP 6576)]
SoapyServerListener::accept([::ffff:192.168.1.86]:52401)
[New Thread 0x747ff050 (LWP 6577)]
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.1.86]:52402)
[New Thread 0x747ff050 (LWP 6578)]
[Thread 0x747ff050 (LWP 6577) exited]
[Thread 0x747ff050 (LWP 6578) exited]
SoapyServerListener::close()
[New Thread 0x747ff050 (LWP 6579)]
[Thread 0x747ff050 (LWP 6579) exited]
[Thread 0x759ff050 (LWP 6575) exited]
SoapyServerListener::close()
[Thread 0x74fff050 (LWP 6576) exited]
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.1.86]:52404)
[New Thread 0x74fff050 (LWP 6580)]
SoapyServerListener::accept([::ffff:192.168.1.86]:52405)
[New Thread 0x759ff050 (LWP 6581)]
[New Thread 0x747ff050 (LWP 6582)]
[Thread 0x747ff050 (LWP 6582) exited]
[Thread 0x74fff050 (LWP 6580) exited]
[Thread 0x759ff050 (LWP 6581) exited]
SoapyServerListener::accept([::ffff:192.168.1.86]:52406)
[New Thread 0x747ff050 (LWP 6583)]
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.1.86]:52407)
[New Thread 0x759ff050 (LWP 6584)]
[New Thread 0x74fff050 (LWP 6585)]
[Thread 0x74fff050 (LWP 6585) exited]
[New Thread 0x74fff050 (LWP 6586)]
[Thread 0x74fff050 (LWP 6586) exited]
[New Thread 0x74fff050 (LWP 6587)]
[New Thread 0x7224f050 (LWP 6588)]
[Thread 0x7224f050 (LWP 6588) exited]
[Thread 0x74fff050 (LWP 6587) exited]
[Thread 0x747ff050 (LWP 6583) exited]
[Thread 0x759ff050 (LWP 6584) exited]
SoapyServerListener::close()
SoapyServerListener::close()

guruofquality, 2018-01-16:
Any backtrace? can you run `bt` in the gdb shell. need to know what call or function died, to look at it closer.

AnsgarSchmidt, 2018-01-16:
Sure let me try it, just a second

AnsgarSchmidt, 2018-01-16:
[New Thread 0x71a4f050 (LWP 803)]
[New Thread 0x710c9050 (LWP 804)]
[New Thread 0x708c9050 (LWP 805)]
[Thread 0x708c9050 (LWP 805) exited]
[New Thread 0x6feff050 (LWP 806)]

Thread 27 "SoapySDRServer" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x710c9050 (LWP 804)]
0x76fba3d8 in memcmp () from /usr/lib/arm-linux-gnueabihf/libarmmem.so
(gdb)

(gdb) bt
#0  0x76fba3d8 in memcmp () from /usr/lib/arm-linux-gnueabihf/libarmmem.so
#1  0x00000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

AnsgarSchmidt, 2018-01-16:
Let me add, the refresh from cubicSDR works fine, the SoapyServer is detected and the SDRplay is identified. SoapyServer crashes as soon as I connect to it.
If I can run more tests, please let me know I am more than happy to help to solve this problem.

guruofquality, 2018-01-17:
I cant tell anything form the backtrace. You probably have to build SoapySDRPlay with some prints and just see where it dies, which function or which mir_sdr call is bombing out, I dont know where that might lead...

If its just dying in connect before any settings are applied, it could be here:
https://github.com/pothosware/SoapySDRPlay/blob/master/Settings.cpp#L33

If you happen to know which call or function on the PC side causes the crash, I can point you to a more specific place.

AnsgarSchmidt, 2018-01-17:
Do you have a version with debug prints I could use and recompile it?

rrobinett, 2018-02-07:
Only when connected by wifi, I am experiencing the same segfaults on both the Pi running SoapyServer and a Win10 or MaxOS CubicSDR running as client.  On the Pi gdb bt shows that I segfaul in memcmp but the stack is hosed, so I can't tell what function is calling memcmp().  I have added a number of std::cerr printouts to SoapyServerThreadData::handlerLoop() which haven't given me any more insight.  I'm a C programmer, not C++, so I'm a little lost as to how to debug further, but it seems that in the context of a Wifi connection there is some unhandled error condition in Soapy code common to the client and server.

guruofquality, 2018-02-08:
@rrobinett This is where all of the calls get handled: https://github.com/pothosware/SoapyRemote/blob/master/server/ClientHandler.cpp#L86 and this is what gets called inside the handler loop.

You can basically see that its getting a enum value "call" and based on that value making the function call into the underlying module. If you can print the value of call (its just an integer), we can figure out which call is actually the one dying and dive deeper and make sure that the stack is really getting into that call and not leaving/segfaulting. Or we find out that the crash is somehow before that and its a socket issue of some kind... Either way, a dump of the call IDs would essentially give a trace of where the client called in and where it last was before something crashed.

Hope that helps, I'm happy to hear where its dying, feel free to post any outputs, etc from this

rrobinett, 2018-02-08:
I added this printout:

bool SoapyClientHandler::handleOnce(SoapyRPCUnpacker &unpacker, SoapyRPCPacker &packer)
{
    SoapyRemoteCalls call;
    unpacker & call;

    std::cerr << "SoapyClientHandler::handleOnce() call =#" << call << std::endl;
    switch (call)
....

And got this logged output which suggests it fails in handling case 708 == SOAPY_REMOTE_GET_GAIN_RANGE_ELEMENT

[New Thread 0x70f3c050 (LWP 5364)]
[New Thread 0x7073c050 (LWP 5365)]
SoapyClientHandler::handleOnce() call =#900
[Thread 0x7073c050 (LWP 5365) exited]
SoapyClientHandler::handleOnce() call =#100
SoapyClientHandler::handleOnce() call =#801
SoapyClientHandler::handleOnce() call =#302
[New Thread 0x6fdff050 (LWP 5366)]
SoapyClientHandler::handleOnce() call =#804
SoapyClientHandler::handleOnce() call =#801
SoapyClientHandler::handleOnce() call =#600
SoapyClientHandler::handleOnce() call =#601
SoapyClientHandler::handleOnce() call =#701
SoapyClientHandler::handleOnce() call =#1402
SoapyClientHandler::handleOnce() call =#1400
SoapyClientHandler::handleOnce() call =#1400
SoapyClientHandler::handleOnce() call =#1400
SoapyClientHandler::handleOnce() call =#1400
SoapyClientHandler::handleOnce() call =#1400
SoapyClientHandler::handleOnce() call =#1400
SoapyClientHandler::handleOnce() call =#1400
SoapyClientHandler::handleOnce() call =#501
SoapyClientHandler::handleOnce() call =#900
SoapyClientHandler::handleOnce() call =#100
SoapyClientHandler::handleOnce() call =#901
SoapyClientHandler::handleOnce() call =#801
SoapyClientHandler::handleOnce() call =#801
SoapyClientHandler::handleOnce() call =#701
SoapyClientHandler::handleOnce() call =#700
SoapyClientHandler::handleOnce() call =#708
SoapyClientHandler::handleOnce() call =#708
SoapyClientHandler::handleOnce() call =#700
SoapyClientHandler::handleOnce() call =#708
SoapyClientHandler::handleOnce() call =#708
SoapyClientHandler::handleOnce() call =#706
SoapyClientHandler::handleOnce() call =#706
SoapyClientHandler::handleOnce() call =#804
SoapyClientHandler::handleOnce() call =#700
SoapyClientHandler::handleOnce() call =#500
SoapyClientHandler::handleOnce() call =#500
SoapyClientHandler::handleOnce() call =#1402
SoapyClientHandler::handleOnce() call =#1401
SoapyClientHandler::handleOnce() call =#1401
SoapyClientHandler::handleOnce() call =#1401
SoapyClientHandler::handleOnce() call =#1401
SoapyClientHandler::handleOnce() call =#1401
SoapyClientHandler::handleOnce() call =#1401
SoapyClientHandler::handleOnce() call =#1401
SoapyClientHandler::handleOnce() call =#902
SoapyClientHandler::handleOnce() call =#700
SoapyClientHandler::handleOnce() call =#708
SoapyClientHandler::handleOnce() call =#708

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x70f3c050 (LWP 5364)]
0x76fbc3a0 in memcmp () from /usr/lib/arm-linux-gnueabihf/libarmmem.so
(gdb)

rrobinett, 2018-02-08:
I spoke too soon.  After adding another printout in case 708, another run terminated after a different case:

SoapyClientHandler::hendleOnce() call =#700
SoapyClientHandler::hendleOnce() call =#708
SoapyClientHandler::hendleOnce() case SOAPY_REMOTE_GET_GAIN_RANGE_ELEMENT
SoapyClientHandler::hendleOnce() call =#708
SoapyClientHandler::hendleOnce() case SOAPY_REMOTE_GET_GAIN_RANGE_ELEMENT
SoapyClientHandler::hendleOnce() call =#700
SoapyClientHandler::hendleOnce() call =#706
SoapyClientHandler::hendleOnce() call =#700
SoapyClientHandler::hendleOnce() call =#706

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x70f3c050 (LWP 5552)]
0x76fbc3cc in memcmp () from /usr/lib/arm-linux-gnueabihf/libarmmem.so
(gdb)

rrobinett, 2018-02-08:
I have discovered that the segfault is associated with higher bitrates.  When I turn the sample rate down to 1 Mbps or less the system starts running stably.  
iperf measures about 35 Mbps across the wifi link and at 1M sample rate Win120 reports 25+ Mbps of network traffic, so Soapy seems to break when packet loss occurs
I can imagine that even in wired configurations the higher sample rates will stimulate a segfault, but for my WSPR application I can run at 1Mbps or less to get around these problems.

guruofquality, 2018-02-12:
Not sure what to make of the crash log other than that, maybe we are changing settings while streaming, not sure if thats an issue for the SDRPlay drivers. If thats the case, we can sprinkle `std::unique_lock <std::mutex> lock(_buf_mutex);` throughout the SoapySDRPlay settings.cpp file.

As far as the high traffic, there's some command line utilities that can simulate the traffic:
```
SoapySDRUtil

  Rate testing options:
    --args[="driver=foo"]                Arguments for testing
    --rate[=stream rate Sps]             Rate in samples per second
    --channels[="0, 1, 2"]               List of channels, default 0
    --direction[=RX or TX]               Specify the channel direction
```

Would be interesting to see if its the traffic since this really doesnt touch settings or multi-threadedness like cubicsdr.

> However it would be good if this error condition were reported by Soapy rather than segfaulting.

Of course thats the problem with segfaults :-) But hopefully its something that can be protected against when its pinned down.

vsonnier, 2018-02-12:
@guruofquality @rrobinett FYI in CubicSDR settings are only _written_ in the unique thread `SDRThread` interleaved in the same thread with `readStream()` calls. 
So, even if the module is not properly multithreaded-safe, there should be no concurrent access between settings and the stream.
On the other hand, settings are indeed _read_ in various threads in the application.

guruofquality, 2018-02-13:
@vsonnier Well, the read settings calls are basically just simple variable reads, so Im not worried about them anyway. But, SoapyRemote client purposely allows for settings and streams to be accessed on separate threads. So even if SoapySDRPlay is accessed sequentially from cubicsdr, the settings and streaming are getting multi-threaded simply because SoapyRemoteServer spawns its own forwarding threads for the streams that are separate from the client settings handling.

So I don't know if this is the problem, but it was intended for settings and streams to be called from different threads, so if SoapySDRPlay's miri API cant handle being accessed simultaneously in different thread contexts, we will need some more of those mutex calls in the settings calls.

@rrobinett It would be interesting to know from the output if the handler completed or bombed out, I would add a print like so in addition to the call code you already have.
```
diff --git a/server/ClientHandler.cpp b/server/ClientHandler.cpp
index 133b1a8..661845e 100644
--- a/server/ClientHandler.cpp
+++ b/server/ClientHandler.cpp
@@ -70,7 +70,9 @@ bool SoapyClientHandler::handleOnce(void)
     bool again = true;
     try
     {
+        printf("Calling handler...\n");
         again = this->handleOnce(unpacker, packer);
+        printf("Finished handler!\n");
     }
     catch (const std::exception &ex)
     {
```

Also, now that we brought it up, there is a second thread to be concerned about: https://github.com/pothosware/SoapyRemote/blob/master/server/ServerStreamData.cpp#L137 `ServerStreamData::sendEndpointWork` calls readStream and sends the data over the socket. It would be interesting to see if its calling into readStream and making it out alive:

```
diff --git a/server/ServerStreamData.cpp b/server/ServerStreamData.cpp
index 063d15d..19b442a 100644
--- a/server/ServerStreamData.cpp
+++ b/server/ServerStreamData.cpp
@@ -173,7 +173,9 @@ void ServerStreamData::sendEndpointWork(void)
         {
             flags = 0; //flags is an in/out parameter and must be cleared for consistency
             const size_t numElems = std::min(mtuElems, elemsLeft);
+            printf("readStream with numelems = %d\n", int(numElems));
             ret = device->readStream(stream, buffs.data(), numElems, flags, timeNs, SOAPY_REMOTE_SOCKET_TIMEOUT_US);
+            printf("readStream returned %d\n", int(ret));
             if (ret == SOAPY_SDR_TIMEOUT) continue;
             if (ret < 0)
             {
```

One of these two threads has to be dying. And once we know which one we can dig into the SoapySDRPlay functions. Although experimenting with some additional mutex calls might be worthwhile as well given the discussion.

vsonnier, 2018-02-15:
@rrobinett @guruofquality On the SoapySDRPlay repo, I've pushed an experimental branch: https://github.com/pothosware/SoapySDRPlay/tree/vso_protected_api_calls that tries to protect against concurrent access the internal state of the module.
@rrobinett Can you try to rebuild this SoapySDRPlay module from that branch and use it together with your server ?

rrobinett, 2018-02-16:
I am very much a newbie here and have other projects preempting my time on this and am pretty unfamiliar with these development environments, so I will be a pretty inefficient diagnostician.
Also, I think you can reproduce my segfaults on a wired LAN if you use 100Mbps LAN and 10 msps  sample rates or force the LAN t0 10 Mbps.

But here is my results this evening:

@guruofquality I added your four printf lines and of course got a lot of output, but the system ran fine a 250K sps.  When I change the sps rate on Cubic from 1 Mbps to 2 Mbps I got an immediate segfault on both the SoapyServer and Cubic.  Here are the last few lines of the Server output:

readStream with numelems = 332
readStream returned 332
readStream with numelems = 332
readStream returned 332
readStream with numelems = 332
readStream returned 332
Calling handler...
SoapyClientHandler::hendleOnce() call =#700
Finished handler!
readStream with numelems = 332
Segmentation fault
pi@raspberrypi:~/Downloads/SoapyRemote-master/server $ 

@vsonnier I downloaded your patch and it seems to contain cpp and hpp files which would replace those in the ...Master/client/ directory.  They compile on the Pi but of course I don't run the client on the Pi.

Unfortunately I have so far only learned how to recompile and debug the SoapyServer on the Pi.   I have been using Win10 as my CubicSDR platform where I have no development environment and also no familiarity with Windows developing.
My Mac is my primary machine where Cubic has also segfaulted,and I have their Xcode IDE but have never used it.

I am willing to debug Cubic and it would seem that doing so on Win10 would be most productive, but I'll need a tutorial on setting up the IDE

vsonnier, 2018-02-16:
@rrobinett Sorry you are wrong, my branch is for a alternate version of the *SoapySDRplay* module, and do not touch SoapyRemote code at all. The server will make use on this SoapySDRplay module to communicate with the device.
So you have to compile the SoapyRemote (what you did until then) and this SoapySDRplay module separately.

Once again Kevin Loughin (@loughkb) scripts are a good inspiration for that. We can safely assume the Mirics API and driver are already installed on you Pi, so which remains is
````bash
git clone https://github.com/pothosware/SoapySDRPlay.git
cd SoapySDRPlay
git checkout -b vso_protected_api_calls origin/vso_protected_api_calls
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Debug
make
make install
````

rrobinett, 2018-02-16:
The 'make install' fails:

pi@raspberrypi:~ $ git clone https://github.com/pothosware/SoapySDRPlay.git
Cloning into 'SoapySDRPlay'...
remote: Counting objects: 361, done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 361 (delta 1), reused 1 (delta 0), pack-reused 355
Receiving objects: 100% (361/361), 104.96 KiB | 0 bytes/s, done.
Resolving deltas: 100% (238/238), done.
Checking connectivity... done.
pi@raspberrypi:~ $ cd SoapySDRPlay/
pi@raspberrypi:~/SoapySDRPlay $ git checkout -b vso_protected_api_calls origin/vso_protected_api_calls
Branch vso_protected_api_calls set up to track remote branch vso_protected_api_calls from origin.
Switched to a new branch 'vso_protected_api_calls'
pi@raspberrypi:~/SoapySDRPlay $ mkdir build
pi@raspberrypi:~/SoapySDRPlay $ cd build
pi@raspberrypi:~/SoapySDRPlay/build $ cmake .. -DCMAKE_BUILD_TYPE=Debug
-- The CXX compiler identification is GNU 4.9.2
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found libsdrplay: /usr/local/include, /usr/local/lib/libmirsdrapi-rsp.so
-- LIBSDRPLAY_INCLUDE_DIRS - /usr/local/include
-- LIBSDRPLAY_LIBRARIES - /usr/local/lib/libmirsdrapi-rsp.so
-- Performing Test HAS_STD_CXX11
-- Performing Test HAS_STD_CXX11 - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/SoapySDRPlay/build
pi@raspberrypi:~/SoapySDRPlay/build $ make
Scanning dependencies of target sdrPlaySupport
[ 25%] Building CXX object CMakeFiles/sdrPlaySupport.dir/Registation.cpp.o
[ 50%] Building CXX object CMakeFiles/sdrPlaySupport.dir/Settings.cpp.o
[ 75%] Building CXX object CMakeFiles/sdrPlaySupport.dir/Streaming.cpp.o
[100%] Linking CXX shared module libsdrPlaySupport.so
[100%] Built target sdrPlaySupport
pi@raspberrypi:~/SoapySDRPlay/build $ make install
[100%] Built target sdrPlaySupport
Install the project...
-- Install configuration: "Debug"
-- Installing: /usr/local/lib/SoapySDR/modules0.6/libsdrPlaySupport.so
CMake Error at cmake_install.cmake:42 (file):
  file INSTALL cannot copy file
  "/home/pi/SoapySDRPlay/build/libsdrPlaySupport.so" to
  "/usr/local/lib/SoapySDR/modules0.6/libsdrPlaySupport.so".

Makefile:104: recipe for target 'install' failed
make: *** [install] Error 1
pi@raspberrypi:~/SoapySDRPlay/build $

SDRplay, 2018-02-16:
looks like a permissions issue, use...

sudo make install

rrobinett, 2018-02-16:
I spoke too soon.  'sudo make install' was successful and there is now a lot more debug output before the segfault:

pi@raspberrypi:~ $ SoapySDRServer --bind >& soapy.log
Segmentation fault
pi@raspberrypi:~ $ tail soapy.log 
mir_sdr_ReadPacket: Gain update confirmed: Gr=47dB GrToggle=0 gset=0xef
grChanged @ 575866
mir_sdr_RSP_SetGr: in: -3 4 0 0
mir_sdr_GpioExtWrite: Addr=19 Data=0xb9
mir_sdr_RSP_SetGr: GR->44[44,0,0,24] gRset->0xEC DCCALmode=4 DCCALspd=0 GrToggle->1
mir_sdr_RSP_SetGr: out: 44 68 4
DCoffsetCorrection: switched to tracking mode diff=1 curr=1769.98 last=1771.00
mir_sdr_ReadPacket: Gain update confirmed: Gr=44dB GrToggle=1 gset=0xec
grChanged @ 576410
DCoffsetCorrection: switched to tracking mode diff=48 curr=2334.23 last=2382.98
pi@raspberrypi:~ $

vsonnier, 2018-02-16:
FYI @rrobinett I updated the branch https://github.com/pothosware/SoapySDRPlay/tree/vso_protected_api_calls to fix a performance regression seen in CubicSDR, although it should change nothing of the present issue.

guruofquality, 2018-02-16:
@vsonnier Something else im not sure about. Obviously the setup/activate stream and anything that talks to the mirics api needs the `_general_state_mutex` to stop other micris calls from happening at the same time. But I think in the case of `SoapySDRPlay::rx_callback`, this is called within the mirics api event thread, so we probably also have to have `_general_state_mutex` locked within the callback so other calls cant take place. And if thats the case, I think some of the other modules like SoapyHackRF/RTL may need the same. Just a thought I had...

vsonnier, 2018-02-16:
@guruofquality Well, in my latest version I think the `rx_callback` protected by `_buf_mutex` should be ok if not absolutely clean: the other fields involved not directly linked to buffer management are `shortsPerWord`, `bufferLength` which are only modified by `setupStream()` so it may be safe *enough*. (cough, cough)

If we start to be super-clean here by adding a `_general_state_mutex` we'll fall into the perf regression I noticed in my previous commit where I added precisely a `_general_state_mutex` in `readStream()` for similar reasons...

As for the other modules indeed none of them have proper concurrent-access protection against their internal state, or even against their own callback threads vs. their internal state either. 

Are we even sure it is a multithreaded problem at all ? My previous commits seem to have little effect on @rrobinett troubles though.

guruofquality, 2018-02-17:
I dont want to get carried away with the threading stuff. It looks like by merit of the implementation, readStream is safe to use from another thread anyway. So we were basically doing the right thing.

Does cubicsdr operate on floats? Because soapyremote will use complex shorts to save bandwidth, so if complex shorts have a bug for SoapySDRPlay, its another reason we would not see it normally. If possible, we can check this by forcing floats over the network with the remote:format stream arg (if cubic can pass this somehow?) https://github.com/pothosware/SoapyRemote/wiki#remote-format

rrobinett did show a crash going into readStream and not coming out:
```
readStream with numelems = 332
Segmentation fault
```

So thats just a hunch but I would try to see how far the prints can get into SoapySDRPlay::readStream(), by sprinkling a few `printf("here %d\n", __LINE__);` around.

Also this probably isnt an issue, but shortsPerWord is not set in the setupStream. I checked that it was initialized, but if the stream was setup once to floats and then again to shorts, it could be set wrong (for whatever thats worth).

vsonnier, 2018-02-17:
> Also this probably isnt an issue, but shortsPerWord is not set in the setupStream. I checked that it was initialized, but if the stream was setup once to floats and then again to shorts, it could be set wrong (for whatever thats worth).

I've seen that and changed it already in previous commits of the test branch :)

CubicSDR indeed operates on CF32, but we can set any allowed format on SoapyRemote.
For example, using the same settings as @rrobinett 's, I tested it on localhost between a SopaySDRServer from the latest PothosSDR win64 distrib and my local CubicSDR.
When starting the connection, we see this:
````
SDR thread starting.
device init()
[INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CF32, scaleFactor=32767, mtu=1400, window=262144)
[INFO] Client side stream bound to 127.0.0.1:60552
[INFO] Client side status bound to 127.0.0.1:60553
[INFO] Using format CS16.
[INFO] Server side stream bound to 127.0.0.1:60554
[INFO] Server side stream connected to 127.0.0.1:60552
[INFO] Server side status connected to 127.0.0.1:60553
[INFO] Configured sender endpoint: dgram=1352 bytes, 332 elements @ 4 bytes, window=256 KiB
[INFO] Client side stream connected to 127.0.0.1:60554
[INFO] Configured receiver endpoint: dgram=1352 bytes, 332 elements @ 4 bytes, window=256 KiB
Device Stream MTU: 332
starting readLoop()
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read blocking...
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
[... 384 lines cut ...]
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(127.0.0.1:62949)
SoapyServerListener::accept(127.0.0.1:62950)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(127.0.0.1:62951)
SoapyServerListener::accept(127.0.0.1:62952)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(127.0.0.1:62953)
SoapyServerListener::accept(127.0.0.1:62954)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(127.0.0.1:62955)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(127.0.0.1:62956)
SoapyServerListener::accept(127.0.0.1:62957)
SoapyServerListener::accept(127.0.0.1:62958)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(127.0.0.1:62959)
````
Of course, no crash seen.

Strykar, 2018-02-26:
@vsonnier Server running Arch arm linux on an Odroid-C2+RSP2, the client is CubicSDR-0.2.3-win64. I tried the vso_protected_api_calls branch and it logs a lot more, so much that systemd muzzles it.

```
Feb 27 01:23:48 Odroid-C2 sudo[16966]: pam_unix(sudo:session): session closed for user root
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: ######################################################
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: ## Soapy Server -- Use any Soapy SDR remotely
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: ######################################################
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: 5a9b14db-f834-1516-8567-4249a8c08002
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: Launching the server... tcp://[::]:55132
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: Server bound to [::]:55132
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: Launching discovery server...
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: Connecting to DNS-SD daemon...
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: [INFO] Avahi version:  avahi 0.7
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: [INFO] Avahi hostname: Odroid-C2
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: [INFO] Avahi domain:   local
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: [INFO] Avahi FQDN:     Odroid-C2.local
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: [INFO] avahi_entry_group_add_service(Odroid-C2._soapy._tcp)
Feb 27 01:23:48 Odroid-C2 SoapySDRServer[16969]: Press Ctrl+C to stop the server
Feb 27 01:23:51 Odroid-C2 sudo[16977]:    alarm : TTY=pts/0 ; PWD=/home/alarm/builds/soapysdrplay-git ; USER=root ; COMMAND=/usr/bin/systemctl status SoapySDRServer
Feb 27 01:23:51 Odroid-C2 sudo[16977]: pam_unix(sudo:session): session opened for user root by alarm(uid=0)
Feb 27 01:23:51 Odroid-C2 sudo[16977]: pam_unix(sudo:session): session closed for user root
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: SoapyServerListener::accept([::ffff:192.168.2.161]:8607)
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: SoapyServerListener::accept([::ffff:192.168.2.161]:8609)
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=3010 rev=0207 serno=1707065E10 bus=001 port=002 devAvail=1
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: SoapyServerListener::accept([::ffff:192.168.2.161]:8610)
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: SoapyServerListener::close()
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: SoapyServerListener::close()
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: SoapyServerListener::accept([::ffff:192.168.2.161]:8611)
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=3010 rev=0207 serno=1707065E10 bus=001 port=002 devAvail=1
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: SoapyServerListener::accept([::ffff:192.168.2.161]:8613)
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: SoapyServerListener::close()
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: SoapyServerListener::accept([::ffff:192.168.2.161]:8614)
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=3010 rev=0207 serno=1707065E10 bus=001 port=002 devAvail=1
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=3010 rev=0207 serno=1707065E10 bus=001 port=002 devAvail=1
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_usb_SetDeviceIdx idx=0 numDevices=1
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: Opened device with idVendor = 0x1df7 idProduct = 0x3010 fwVersion = 0x0207 busNum = 001 portNum = 002
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_Init: starting hardware initialization
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_Init: gR=60dB fs=2.000MHz rf=98.800MHz bw=1.536MHz if=0.000MHz
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: DownConvert: Enable=0 DecM=1 OutScale=0 (fs=2.000000 bw=1536 if=0)
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: initState: numActivePackets=4
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_usb_USB DLL: Revision 0.1.1
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_2500_Init: fnaddr = 7
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_2500_Init: adjusting squelch trim 0x1, rx gating enable 1, tx_trim 0, reg2 = 0x4801
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: initHw: Register7 = 0x000014
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: initRfFreqDependentHw(1): Tuner Register0 = 0x04f420
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_GpioExtWrite: Addr=0 Data=0x00
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_GpioExtWrite: Addr=1 Data=0x00
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_GpioExtWrite: Addr=18 Data=0x5d
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_GpioExtWrite: Addr=19 Data=0xa5
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: initRfFreqDependentHw(1): ExtGpioA = 0x5d ExtGpioB = 0xa5
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_SetFs: Sample Freq requested 2000000.000000
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_SetFs: Fs->FsNomHz+dFsHz=2000000.0+0.0Hz=2000000.0Hz FsToggle->1
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_SetRf: f->98800000.000Hz (int=20 frac=af0 afc=0) fSynth:3161600000.000
Feb 27 01:25:05 Odroid-C2 SoapySDRServer[16969]: mir_sdr_SetRf: Rf->RfNomHz+dRfHz+LifHz+Lif1Hz=98800000.0+0.0+0.0Hz+0.0Hz=98800000.0Hz RfToggle->1
[... cut ...]
-- /etc/systemd/journald.conf. See journald.conf(5) for details.
Feb 27 01:25:35 Odroid-C2 SoapySDRServer[16969]: Gap=0xffffff04 e=0xfc a=0x0
Feb 27 01:25:35 Odroid-C2 SoapySDRServer[16969]: Gap=0xffffff04 e=0xfc a=0x0
Feb 27 01:25:35 Odroid-C2 SoapySDRServer[16969]: Gap=0xffffff04 e=0xfc a=0x0
Feb 27 01:25:35 Odroid-C2 SoapySDRServer[16969]: ate _currentBuff = 0x0x7f8ae7c300
```

Let me know what else I can do to help troubleshoot this, CubicSDR sees the RSP2, but it doesn't show a waterfall, like nothing's connected - https://imgur.com/a/W4bcq

vsonnier, 2018-02-28:
@Strykar Thanks for giving us a hand by testing that branch ! I'm not an expert on Linux matters, but why the usual `printf()` are going to some kind of system-wide journal there ? I don't see anything on Google about that.
Anyway, as `man journald.conf` says, modifies the following entries in `journald.conf` as the following:
-  `RateLimitInterval=0` 
-  `RateLimitBurst=0`
- `Storage=auto` or `Storage=persistent` so that they are saved on disk.

SDRplay, 2018-03-04:
I think I've got to the bottom of the issue on Odroid.

For anyone that has Odroid, insert...

mir_sdr_SetTransferMode(mir_sdr_BULK);

immediately before the mir_sdr_StreamInit call in the Streaming.cpp file and rebuild SoapySDRPlay.

It's a compilation issue and I'll fix it in the API build as soon as I can.

vsonnier, 2018-03-04:
@SDRplay Do you mean the trace spam issue or the real crash one ?

SDRplay, 2018-03-04:
the real crash one - the implementation of isochronous mode isn't good enough :-) on ARM devices, so we need to default to bulk mode. We use __arm__ define to check whether we're building on an ARM device, which I didn't realise only works on 32bit ARM devices. If you want to check for 64bit ARM devices you also need to use __aarch64__ - nothing like consistency eh?

SDRplay, 2018-03-04:
This also found a flaw in our testing/release procedure, so we'll tighten that up. If the fix works, then I would recommend turning off debug `mir_sdr_DebugEnable(0);` to avoid performance issues from the API

vsonnier, 2018-03-04:
@SDRplay woot ! That should solve Rasberry PI 2 issues as well ?

SDRplay, 2018-03-04:
RPi2 is 32bit I believe so as long as the RPi version is downloaded from our website then I'm not sure it's the same issue. I've had someone post on our forum that they have the same issue with our RPi3 image using SoapyRemote, so I've asked for more debug info and will report if I find anything.

I think what I need to do is to just have a 32bit ARM and a 64bit ARM build of the API in a single installer, and then the installer to check what the processor is and put the right one in place. We have people installing the wrong API sometimes and the installation script doesn't do a good enough job of checking the processor to be honest.

What I'm really wondering is whether I need to host a repository and so the API would be able to be installed by something like...

sudo apt-get install SDRplayAPI

Maybe that just solves the whole problem, or creates a new one :-) the Linux delivery system is baffling to me at times!

vsonnier, 2018-03-04:
@SDRplay yes, I meant indeed Rasberry PI 3 which is 64 bit. The reported issues on CubicSDR base are also about Pi3 so consistent with the bug.

SDRplay, 2018-03-04:
temporarily can you add the transfer mode statement into the SoapySDRPlay library based on a runtime check for ARM (either 32 or 64 bit) until I can get the API/installer issue resolved?

I can probably do it if you don't have time.

vsonnier, 2018-03-04:
OK I'll do it. With which #define I need to guard the code ?

SDRplay, 2018-03-04:
```
#if defined(__arm__) || defined(__aarch64__)
mir_sdr_SetTransferMode(mir_sdr_BULK);
#endif
```

should do it

SDRplay, 2018-03-04:
needs to go immediately before StreamInit and ideally to be tested :-)

SDRplay, 2018-03-04:
sorry, I meant compile time, not runtime! too many things going on at once!

vsonnier, 2018-03-04:
We need a chat :)

vsonnier, 2018-03-04:
Done.

vsonnier, 2018-03-04:
@rrobinett We may have solved the issue so you can now try the master branch:
````bash
git clone https://github.com/pothosware/SoapySDRPlay.git
cd SoapySDRPlay
git checkout -b master origin/master
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Debug
make
sudo make install
````

SDRplay, 2018-03-04:
does the Debug build type enable the API debug output?

It will affect performance, but is good if there are still any issues.

vsonnier, 2018-03-04:
@SDRplay I wouldn't know, I use CMake-Gui on Windows to configure. I use "Release" and "Release with debug info" configuration exclusively and I don't know if it applies on the other platforms.

rrobinett, 2018-03-04:
Sorry that I've been too busy to test, but I will try in the next several days.  Thanks

Strykar, 2018-03-05:
@SDRplay fix works! CubicSDR can now receive data, though anything over 1 Mhz seems to be too much.

SDRplay, 2018-03-05:
If you've still got debug enabled that will be a problem. Also, 16bit I/Q data will be quite bandwidth hungry at higher sample rates. If the library supports it, you might try 8 bit mode. May @vsonnier can confirm whether that is supported. To be honest remote data is really only suitable for small bandwidths on any system.

Strykar, 2018-03-05:
@SDRplay Debug is disabled in head now, so it isn't enabled. I have a 1Gbit Ethernet connection, how much bandwidth do we need? Is this an issue of network bandwidth or small SBC's being unable to stuff that bandwidth without dedicated PHY's like RPi?

SDRplay, 2018-03-05:
Most SBCs like the Raspberry Pi have an inherent problem because ALL of the peripheral data is shared on the USB bus including the network port. I'm not sure about the Odroid. I've backed a new SBC on kickstarter (https://www.kickstarter.com/projects/librecomputer/libre-computer-board-tritium-sbc-linux-android-7-n) that doesn't have this shared bandwidth which I hope to receive in the next couple of months and I'll be able to gather some data on throughput. You'll also need to consider the processing requirement and the latency huge sample rates will create. It may be worth seeing if there are any options you can use in SoapyRemote to change the CPU vs bandwidth - maybe @guruofquality or @vsonnier can comment? What sort of sample rate were you expecting?

Strykar, 2018-03-05:
The Odroid has a dedicated PHY and does not share the bus with USB devices, it's one of the reasons I chose it, along with boatloads more CPU power over the RPi. I can understand latency being an issue, and am happy to work within these limits.
TBH, looking at https://libre.computer/products/boards/all-h3-cc/ I don't see how this is faster than an Odroid, it even has just 100 Mbit LAN?

Speaking of CPU, I'm awaiting delivery of an all Aluminum case for the Odroid, perhaps that can help me thermal throttling running GQRX and accessing it via remote X. If not, this might a better option: https://www.notebookcheck.net/Benchmarks-for-the-Intel-Hades-Canyon-NUC-with-Core-i7-8809G-CPU-and-RX-Vega-M-GPU-leak-out.287220.0.html

vsonnier, 2018-03-05:
@Strykar you can try CS16 format and tuning the MTU among other things:
[image]

rrobinett, 2018-03-20:
I am finally able to resume work on this.

Because my original Pi environment was corrupted I flashed the latest raspberian and am trying to cleanly build Soapy following the instructions of @vsonnier, but I get errors, first in:

cmake .. -DCMAKE_BUILD_TYPE=Debug

Then when I execute cmake :

### pi@raspberrypi:~/SoapySDRPlay $ git checkout -b master origin/master
fatal: A branch named 'master' already exists.
pi@raspberrypi:~/SoapySDRPlay $ mkdir build
pi@raspberrypi:~/SoapySDRPlay $ cmake .. -DCMAKE_BUILD_TYPE=Debug
CMake Error: The source directory "/home/pi" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
pi@raspberrypi:~/SoapySDRPlay $ cd build/
pi@raspberrypi:~/SoapySDRPlay/build $ cmake .. -DCMAKE_BUILD_TYPE=Debug
-- The CXX compiler identification is GNU 6.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:7 (find_package):
  Could not find a package configuration file provided by "SoapySDR"
  (requested version 0.4.0) with any of the following names:

    SoapySDRConfig.cmake
    soapysdr-config.cmake

  Add the installation prefix of "SoapySDR" to CMAKE_PREFIX_PATH or set
  "SoapySDR_DIR" to a directory containing one of the above files.  If
  "SoapySDR" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/pi/SoapySDRPlay/build/CMakeFiles/CMakeOutput.log".
pi@raspberrypi:~/SoapySDRPlay/build $

What am I doing wrong?

guruofquality, 2018-03-20:
It cant find SoapySDR development files. Did you install SoapySDR? if so can you share the output of make install?

rrobinett, 2018-03-20:
No.
I confess I am lost in the web site looking for instructions on how to set
up this whole development environment

rrobinett, 2018-03-20:
So here is the output:

pi@raspberrypi:~/SoapySDR-master/bulid $ sudo make install
[ 68%] Built target SoapySDR
[ 81%] Built target SoapySDRUtil
[ 87%] Built target TestTimeConversion
[ 93%] Built target TestFormatParser
[100%] Built target TestKwargsMarkup
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/share/cmake/SoapySDR/SoapySDRConfig.cmake
-- Installing: /usr/local/share/cmake/SoapySDR/SoapySDRUtil.cmake
-- Installing: /usr/local/share/cmake/SoapySDR/SoapySDRConfigVersion.cmake
-- Installing: /usr/local/include/SoapySDR
-- Installing: /usr/local/include/SoapySDR/Device.h
-- Installing: /usr/local/include/SoapySDR/Version.hpp
-- Installing: /usr/local/include/SoapySDR/ConverterPrimatives.hpp
-- Installing: /usr/local/include/SoapySDR/Errors.h
-- Installing: /usr/local/include/SoapySDR/Logger.hpp
-- Installing: /usr/local/include/SoapySDR/Time.hpp
-- Installing: /usr/local/include/SoapySDR/Types.hpp
-- Installing: /usr/local/include/SoapySDR/Modules.hpp
-- Installing: /usr/local/include/SoapySDR/ConverterRegistry.hpp
-- Installing: /usr/local/include/SoapySDR/Config.h
-- Installing: /usr/local/include/SoapySDR/Registry.hpp
-- Installing: /usr/local/include/SoapySDR/Formats.hpp
-- Installing: /usr/local/include/SoapySDR/Config.hpp
-- Installing: /usr/local/include/SoapySDR/Errors.hpp
-- Installing: /usr/local/include/SoapySDR/Modules.h
-- Installing: /usr/local/include/SoapySDR/Types.h
-- Installing: /usr/local/include/SoapySDR/Time.h
-- Installing: /usr/local/include/SoapySDR/Device.hpp
-- Installing: /usr/local/include/SoapySDR/Version.h
-- Installing: /usr/local/include/SoapySDR/Logger.h
-- Installing: /usr/local/include/SoapySDR/Constants.h
-- Installing: /usr/local/include/SoapySDR/Formats.h
-- Installing: /usr/local/lib/libSoapySDR.so.0.7.0
-- Installing: /usr/local/lib/libSoapySDR.so.0.7
-- Installing: /usr/local/lib/libSoapySDR.so
-- Installing: /usr/local/lib/pkgconfig/SoapySDR.pc
-- Installing: /usr/local/bin/SoapySDRUtil
-- Set runtime path of "/usr/local/bin/SoapySDRUtil" to ""
-- Installing: /usr/local/share/man/man1/SoapySDRUtil.1
pi@raspberrypi:~/SoapySDR-master/bulid $ SoapySDRUtil --info
SoapySDRUtil: error while loading shared libraries: libSoapySDR.so.0.7: cannot open shared object file: No such file or directory
pi@raspberrypi:~/SoapySDR-master/bulid $ sudo ldconfig
pi@raspberrypi:~/SoapySDR-master/bulid $ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.0-unknown
API Version: v0.7.0
ABI Version: v0.7
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.7
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
pi@raspberrypi:~/SoapySDR-master/bulid $

vsonnier, 2018-03-20:
@rrobinett The problem comes from 

> fatal: A branch named 'master' already exists.

Delete all the SoapySDR , SoapySDRPlay, SoapyRemote directories, and do the following:
````
git clone https://github.com/pothosware/SoapySDR.git
git clone https://github.com/pothosware/SoapySDRPlay.git
git clone https://github.com/pothosware/SoapyRemote.git
cd SoapySDR
mkdir build
cd build
cmake ..
make -j4
make install
ldconfig
cd ..
cd ..
# Now we build the SDRPlay module for Soapy
cd SoapySDRPlay
mkdir build
cd build
cmake ..
make
make install
cd ..
cd ..
# And we build SoapyRemote
cd SoapyRemote
mkdir build
cd build
#cmake ..
cmake .. -DCMAKE_BUILD_TYPE=Release
make
make install
cd ..
cd ..
````

(From @loughkb scripts)

[4 more replies]

closed 2018-03-21 by guruofquality