pothosware/SoapyRemote #32: soapysdrserver segmentation fault with hackrf

Vertikar, 2017-12-04:
Seem to be able to get this to happen quite frequently.
Running SoapySDR on Ubuntu 16.04
Running Pothos GUI on Windows, this is the main error I could find on the windows side:

`11:21:45.894707] PothosGui.EnvironmentEval: zone[]: I/O error: RemoteProxyEnvironment::transact(): connection inactive - Remote environment crashed`

Output on the linux side:
```
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.0.180]:20717)
Segmentation fault (core dumped)
```

And output after starting strace and replicating:

```
select(4, [3], NULL, NULL, {0, 50000})  = 0 (Timeout)
select(4, [3], NULL, NULL, {0, 50000})  = 0 (Timeout)
select(4, [3], NULL, NULL, {0, 50000})  = 0 (Timeout)
select(4, [3], NULL, NULL, {0, 50000})  = 0 (Timeout)
select(4, [3], NULL, NULL, {0, 50000})  = 1 (in [3], left {0, 12507})
accept(3, {sa_family=AF_INET6, sin6_port=htons(20776), inet_pton(AF_INET6, "::ffff:192.168.0.180", &sin6_addr), sin6_flowinfo=0, sin6_scope_id=0}, [28]) = 6
setsockopt(6, SOL_TCP, TCP_NODELAY, [1], 4) = 0
setsockopt(6, SOL_TCP, TCP_QUICKACK, [1], 4) = 0
getpeername(6, {sa_family=AF_INET6, sin6_port=htons(20776), inet_pton(AF_INET6, "::ffff:192.168.0.180", &sin6_addr), sin6_flowinfo=0, sin6_scope_id=0}, [28]) = 0
write(1, "SoapyServerListener::accept([::f"..., 58SoapyServerListener::accept([::ffff:192.168.0.180]:20776)
) = 58
clone(child_stack=0x7f436cd84bb0, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tidptr=0x7f436cd859d0, tls=0x7f436cd85700, child_tidptr=0x7f436cd859d0) = 28370
select(4, [3], NULL, NULL, {0, 50000})  = 1 (in [3], left {0, 41046})
accept(3, {sa_family=AF_INET6, sin6_port=htons(20777), inet_pton(AF_INET6, "::ffff:192.168.0.180", &sin6_addr), sin6_flowinfo=0, sin6_scope_id=0}, [28]) = 7
setsockopt(7, SOL_TCP, TCP_NODELAY, [1], 4) = 0
setsockopt(7, SOL_TCP, TCP_QUICKACK, [1], 4) = 0
getpeername(7, {sa_family=AF_INET6, sin6_port=htons(20777), inet_pton(AF_INET6, "::ffff:192.168.0.180", &sin6_addr), sin6_flowinfo=0, sin6_scope_id=0}, [28]) = 0
write(1, "SoapyServerListener::accept([::f"..., 58SoapyServerListener::accept([::ffff:192.168.0.180]:20777)
) = 58
clone(child_stack=0x7f436e430bb0, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tidptr=0x7f436e4319d0, tls=0x7f436e431700, child_tidptr=0x7f436e4319d0) = 28371
select(4, [3], NULL, NULL, {0, 50000})  = 1 (in [3], left {0, 33794})
accept(3, {sa_family=AF_INET6, sin6_port=htons(20778), inet_pton(AF_INET6, "::ffff:192.168.0.180", &sin6_addr), sin6_flowinfo=0, sin6_scope_id=0}, [28]) = 8
setsockopt(8, SOL_TCP, TCP_NODELAY, [1], 4) = 0
setsockopt(8, SOL_TCP, TCP_QUICKACK, [1], 4) = 0
getpeername(8, {sa_family=AF_INET6, sin6_port=htons(20778), inet_pton(AF_INET6, "::ffff:192.168.0.180", &sin6_addr), sin6_flowinfo=0, sin6_scope_id=0}, [28]) = 0
write(1, "SoapyServerListener::accept([::f"..., 58SoapyServerListener::accept([::ffff:192.168.0.180]:20778)
) = 58
clone(child_stack=0x7f436ec31bb0, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tidptr=0x7f436ec329d0, tls=0x7f436ec32700, child_tidptr=0x7f436ec329d0) = 28373
write(1, "SoapyServerListener::close()\n", 29SoapyServerListener::close()
) = 29
close(6)                                = 0
write(1, "SoapyServerListener::close()\n", 29SoapyServerListener::close()
) = 29
close(7)                                = 0
select(4, [3], NULL, NULL, {0, 50000})  = 1 (in [3], left {0, 44207})
accept(3, {sa_family=AF_INET6, sin6_port=htons(20779), inet_pton(AF_INET6, "::ffff:192.168.0.180", &sin6_addr), sin6_flowinfo=0, sin6_scope_id=0}, [28]) = 6
setsockopt(6, SOL_TCP, TCP_NODELAY, [1], 4) = 0
setsockopt(6, SOL_TCP, TCP_QUICKACK, [1], 4) = 0
getpeername(6, {sa_family=AF_INET6, sin6_port=htons(20779), inet_pton(AF_INET6, "::ffff:192.168.0.180", &sin6_addr), sin6_flowinfo=0, sin6_scope_id=0}, [28]) = 0
write(1, "SoapyServerListener::accept([::f"..., 58SoapyServerListener::accept([::ffff:192.168.0.180]:20779)
) = 58
clone(child_stack=0x7f436e430bb0, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tidptr=0x7f436e4319d0, tls=0x7f436e431700, child_tidptr=0x7f436e4319d0) = 28374
select(4, [3], NULL, NULL, {0, 50000} <unfinished ...>
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)
karl@mini:~$
```

Vertikar, 2017-12-04:
Saw in the other segfault issue that you asked for gdb to run, here's the last few lines of output from that:

```
[Thread 0x7fffe7991700 (LWP 28829) exited]
[Thread 0x7ffff50fe700 (LWP 28826) exited]
[Thread 0x7ffff58ff700 (LWP 28827) exited]
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.0.180]:21313)
[New Thread 0x7ffff58ff700 (LWP 28830)]
SoapyServerListener::accept([::ffff:192.168.0.180]:21314)
[New Thread 0x7ffff50fe700 (LWP 28831)]
[New Thread 0x7fffe7991700 (LWP 28832)]
[Thread 0x7fffe7991700 (LWP 28832) exited]
[Thread 0x7ffff58ff700 (LWP 28830) exited]
[Thread 0x7ffff50fe700 (LWP 28831) exited]
SoapyServerListener::accept([::ffff:192.168.0.180]:21315)
[New Thread 0x7fffe7991700 (LWP 28833)]
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.0.180]:21316)
[New Thread 0x7ffff50fe700 (LWP 28834)]
[New Thread 0x7ffff58ff700 (LWP 28835)]
[Thread 0x7ffff58ff700 (LWP 28835) exited]
[New Thread 0x7ffff58ff700 (LWP 28836)]
[New Thread 0x7fffe7190700 (LWP 28837)]
[New Thread 0x7fffe698f700 (LWP 28838)]
[Thread 0x7fffe698f700 (LWP 28838) exited]
[New Thread 0x7fffe618e700 (LWP 28839)]

Thread 39 "SoapySDRServer" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe7991700 (LWP 28833)]
0x00007ffff6d66cc0 in _IO_vfprintf_internal (s=s@entry=0x7fffe798fcc0, format=<optimised out>, format@entry=0x7ffff46f9520 "hackrf_set_freq(%f) returned %s", ap=ap@entry=0x7fffe798fe10) at vfprintf.c:1632
1632    vfprintf.c: No such file or directory.
(gdb)
```

guruofquality, 2018-01-17:
Just like this issue: https://github.com/pothosware/SoapyRemote/issues/26 I really need either the perfect traceback or some good debug prints from either the server side or the client, to be of any use debugging from github like this. I have a hunch that SoapyRemote itself is pretty stable, but some of the hardware support modules could have the unexpected bug. So I suspect its a bug in the SoapyHackrf for example, one of the function calls in there, or lower like in libhackrf. Just a guess...

guruofquality, 2018-08-25:
Closing, I just need more info. One of the bugs we have found for other devices is that dangling streams are closed but not deactivated. SoapyHackrf may need to shutdown any potential threads in close stream. This may or may not be related -- Just stating.

closed 2018-08-25 by guruofquality