pothosware/SoapyRemote #35: Cmke ..     error installing Soapy

santiaarana, 2018-02-06:
I have error installaing SoapyRemote. 
Raspberry Pi 3 Model B. 
Raspbian GNU/Linux 9.3 (stretch)

Can you help me please?

pi@raspberrypi:~/SoapyRemote/build$ cmake ..
-- Build type not specified: defaulting to release.
CMake Warning at CMakeLists.txt:15 (find_package):
  Could not find a package configuration file provided by "SoapySDR"
  (requested version 0.4.0) with any of the following names:

    SoapySDRConfig.cmake
    soapysdr-config.cmake

  Add the installation prefix of "SoapySDR" to CMAKE_PREFIX_PATH or set
  "SoapySDR_DIR" to a directory containing one of the above files.  If
  "SoapySDR" provides a separate development package or SDK, be sure it has
  been installed.

CMake Error at client/CMakeLists.txt:4 (SOAPY_SDR_MODULE_UTIL):
  Unknown CMake command "SOAPY_SDR_MODULE_UTIL".

guruofquality, 2018-02-06:
Is SoapySDR installed? Its a dependency: https://github.com/pothosware/SoapyRemote/wiki#dependencies

santiaarana, 2018-02-07:
Thanks. I just needed to read a little more!!!
"The remote device support has no additional dependencies other than SoapySDR itself"
I installed soapyremote without further problems. But I plug my RSP1A and ...

pi@raspberrypi:~$ SoapySDRUtil --find
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]
No devices found!

No devices found I can start the server, but without a receiver. Can you help me again?

pi@raspberrypi:~$ SoapySDRServer --bind=192.168.0.105:1234
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

542d0ee0-2ca5-1511-8567-0399007f0101
Launching the server... tcp://192.168.0.105:1234
Server bound to 192.168.0.105:1234
Launching discovery server...
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]
Press Ctrl+C to stop the server

SDRplay, 2018-02-07:
You do have SoapySDRPlay installed? If not, remember you'll also need the RSP API first which you can get from our downloads page (https://www.sdrplay.com/downloads)

guruofquality, 2018-02-08:
On the server:
* install soapysdr
* install soapyremote
* install soapysdrplay (or other hardware support)
* `SoapySDRUtil` --info should display remote and sdrplay drivers
* `SoapySDRUtil --find` should show the devices like sdrplay
* SoapySDRServer --bind

On the client:
* install soapysdr
* install soapyremote
* `SoapySDRUtil --find` should show the devices over the remote connection

santiaarana, 2018-02-08:
Thank you very much for your help. I have been able to install Soapyremote. Now I learn with GQRX SDR. It is the only sdr client that works for me.
Thanks again!

closed 2018-03-21 by guruofquality