pothosware/SoapyRemote #37: Cross-compile attempt

unixpunk, 2018-02-23:
First off, thanks for the work on this project, it should perfectly fit the bill for what I'm attempting.  :)  I'm looking for SoapyRemote to cross-compile for armv7,neon and I've gotten most of the way there.  Once compiled, this would then be able to run ON a PlutoSDR, turning it into a fully remote SDR as it already support OTG eth and wifi out of the box.  I've been successful with SoapySDR, SoapyPlutoSDR and rx_tools for Openwebrx so far.  Any help or guidance is appreciated.  During make, I'm getting the following:

``[ 41%] Built target SoapySDRRemoteCommon
Scanning dependencies of target remoteSupport
[ 45%] Building CXX object client/CMakeFiles/remoteSupport.dir/Registration.cpp.o
[ 50%] Building CXX object client/CMakeFiles/remoteSupport.dir/Settings.cpp.o
[ 54%] Building CXX object client/CMakeFiles/remoteSupport.dir/Streaming.cpp.o
[ 58%] Building CXX object client/CMakeFiles/remoteSupport.dir/LogAcceptor.cpp.o
[ 62%] Building CXX object client/CMakeFiles/remoteSupport.dir/ClientStreamData.cpp.o
[ 66%] Building CXX object client/CMakeFiles/remoteSupport.dir/Version.cpp.o
[ 70%] Linking CXX shared module libremoteSupport.so
CMakeFiles/remoteSupport.dir/Settings.cpp.o: In function `getGainRange':
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Settings.cpp:544: undefined reference to `SoapySDR::Range::Range()'
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Settings.cpp:560: undefined reference to `SoapySDR::Range::Range()'
CMakeFiles/remoteSupport.dir/Settings.cpp.o: In function `~SoapyRemoteDevice':
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Settings.cpp:48: undefined reference to `SoapySDR::Device::~Device()'
CMakeFiles/remoteSupport.dir/Settings.cpp.o: In function `getSensorInfo':
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Settings.cpp:1011: undefined reference to `SoapySDR::ArgInfo::ArgInfo()'
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Settings.cpp:1056: undefined reference to `SoapySDR::ArgInfo::ArgInfo()'
CMakeFiles/remoteSupport.dir/Settings.cpp.o: In function `SoapyRemoteDevice':
/opt/Xilinx/SDK/2016.4/gnu/arm/lin/arm-xilinx-linux-gnueabi/include/c++/4.9.2/bits/basic_string.h:240: undefined reference to `SoapySDR::Device::~Device()'
CMakeFiles/remoteSupport.dir/Settings.cpp.o: In function `logf':
/home/steve/Desktop/SDR/Pluto/ready/SoapySDR/include/SoapySDR/Logger.hpp:47: undefined reference to `SoapySDR::vlogf(SoapySDRLogLevel, char const*, std::__va_list)'
CMakeFiles/remoteSupport.dir/Settings.cpp.o:(.data.rel.ro+0x1d4): undefined reference to `typeinfo for SoapySDR::Device'
CMakeFiles/remoteSupport.dir/Streaming.cpp.o: In function `logf':
/home/steve/Desktop/SDR/Pluto/ready/SoapySDR/include/SoapySDR/Logger.hpp:47: undefined reference to `SoapySDR::vlogf(SoapySDRLogLevel, char const*, std::__va_list)'
CMakeFiles/remoteSupport.dir/Streaming.cpp.o: In function `getStreamArgsInfo':
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Streaming.cpp:78: undefined reference to `SoapySDR::ArgInfo::ArgInfo()'
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Streaming.cpp:87: undefined reference to `SoapySDR::ArgInfo::ArgInfo()'
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Streaming.cpp:95: undefined reference to `SoapySDR::ArgInfo::ArgInfo()'
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Streaming.cpp:104: undefined reference to `SoapySDR::ArgInfo::ArgInfo()'
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Streaming.cpp:113: undefined reference to `SoapySDR::ArgInfo::ArgInfo()'
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Streaming.cpp:119: undefined reference to `SoapySDR::Range::Range(double, double, double)'
CMakeFiles/remoteSupport.dir/Streaming.cpp.o: In function `setupStream':
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Streaming.cpp:157: undefined reference to `SoapySDR::formatToSize(std::string const&)'
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Streaming.cpp:298: undefined reference to `SoapySDR::formatToSize(std::string const&)'
CMakeFiles/remoteSupport.dir/LogAcceptor.cpp.o: In function `logf':
/home/steve/Desktop/SDR/Pluto/ready/SoapySDR/include/SoapySDR/Logger.hpp:47: undefined reference to `SoapySDR::vlogf(SoapySDRLogLevel, char const*, std::__va_list)'
CMakeFiles/remoteSupport.dir/LogAcceptor.cpp.o: In function `handlerLoop':
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/LogAcceptor.cpp:119: undefined reference to `SoapySDR::log(SoapySDRLogLevel, std::string const&)'
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/LogAcceptor.cpp:119: undefined reference to `SoapySDR::log(SoapySDRLogLevel, std::string const&)'
../common/libSoapySDRRemoteCommon.a(SoapyRPCUnpacker.cpp.o): In function `logf':
/home/steve/Desktop/SDR/Pluto/ready/SoapySDR/include/SoapySDR/Logger.hpp:47: undefined reference to `SoapySDR::vlogf(SoapySDRLogLevel, char const*, std::__va_list)'
../common/libSoapySDRRemoteCommon.a(SoapyRPCUnpacker.cpp.o): In function `operator&':
/home/steve/Desktop/SDR/Pluto/SoapyRemote/common/SoapyRPCUnpacker.cpp:233: undefined reference to `SoapySDR::Range::Range(double, double, double)'
../common/libSoapySDRRemoteCommon.a(SoapyRPCUnpacker.cpp.o): In function `_Construct<SoapySDR::Range>':
/opt/Xilinx/SDK/2016.4/gnu/arm/lin/arm-xilinx-linux-gnueabi/include/c++/4.9.2/bits/stl_construct.h:75: undefined reference to `SoapySDR::Range::Range()'
/opt/Xilinx/SDK/2016.4/gnu/arm/lin/arm-xilinx-linux-gnueabi/include/c++/4.9.2/bits/stl_construct.h:75: undefined reference to `SoapySDR::Range::Range()'
../common/libSoapySDRRemoteCommon.a(SoapyRPCUnpacker.cpp.o): In function `_Construct<SoapySDR::ArgInfo>':
/opt/Xilinx/SDK/2016.4/gnu/arm/lin/arm-xilinx-linux-gnueabi/include/c++/4.9.2/bits/stl_construct.h:75: undefined reference to `SoapySDR::ArgInfo::ArgInfo()'
/opt/Xilinx/SDK/2016.4/gnu/arm/lin/arm-xilinx-linux-gnueabi/include/c++/4.9.2/bits/stl_construct.h:75: undefined reference to `SoapySDR::ArgInfo::ArgInfo()'
../common/libSoapySDRRemoteCommon.a(SoapySSDPEndpoint.cpp.o): In function `logf':
/home/steve/Desktop/SDR/Pluto/ready/SoapySDR/include/SoapySDR/Logger.hpp:47: undefined reference to `SoapySDR::vlogf(SoapySDRLogLevel, char const*, std::__va_list)'
../common/libSoapySDRRemoteCommon.a(SoapyMDNSEndpointAvahi.cpp.o): In function `logf':
/home/steve/Desktop/SDR/Pluto/ready/SoapySDR/include/SoapySDR/Logger.hpp:47: undefined reference to `SoapySDR::vlogf(SoapySDRLogLevel, char const*, std::__va_list)'
CMakeFiles/remoteSupport.dir/Registration.cpp.o: In function `logf':
/home/steve/Desktop/SDR/Pluto/ready/SoapySDR/include/SoapySDR/Logger.hpp:47: undefined reference to `SoapySDR::vlogf(SoapySDRLogLevel, char const*, std::__va_list)'
CMakeFiles/remoteSupport.dir/Registration.cpp.o: In function `__static_initialization_and_destruction_0':
/home/steve/Desktop/SDR/Pluto/SoapyRemote/client/Registration.cpp:238: undefined reference to `SoapySDR::Registry::Registry(std::string const&, std::vector<std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > >, std::allocator<std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > > > (* const&)(std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&), SoapySDR::Device* (* const&)(std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&), std::string const&)'
CMakeFiles/remoteSupport.dir/Registration.cpp.o: In function `_M_dispose':
/opt/Xilinx/SDK/2016.4/gnu/arm/lin/arm-xilinx-linux-gnueabi/include/c++/4.9.2/bits/basic_string.h:240: undefined reference to `SoapySDR::Registry::~Registry()'
CMakeFiles/remoteSupport.dir/Version.cpp.o: In function `__static_initialization_and_destruction_0':
/home/steve/Desktop/SDR/Pluto/SoapyRemote/build/client/Version.cpp:2: undefined reference to `SoapySDR::ModuleVersion::ModuleVersion(std::string const&)'
../common/libSoapySDRRemoteCommon.a(SoapyRPCSocket.cpp.o): In function `logf':
/home/steve/Desktop/SDR/Pluto/ready/SoapySDR/include/SoapySDR/Logger.hpp:47: undefined reference to `SoapySDR::vlogf(SoapySDRLogLevel, char const*, std::__va_list)'
../common/libSoapySDRRemoteCommon.a(SoapyStreamEndpoint.cpp.o): In function `logf':
/home/steve/Desktop/SDR/Pluto/ready/SoapySDR/include/SoapySDR/Logger.hpp:47: undefined reference to `SoapySDR::vlogf(SoapySDRLogLevel, char const*, std::__va_list)'
[... cut ...]
/home/steve/Desktop/SDR/Pluto/SoapyRemote/common/SoapyStreamEndpoint.cpp:227: undefined reference to `SoapySDR::log(SoapySDRLogLevel, std::string const&)'
collect2: error: ld returned 1 exit status
client/CMakeFiles/remoteSupport.dir/build.make:227: recipe for target 'client/libremoteSupport.so' failed
make[2]: *** [client/libremoteSupport.so] Error 1
CMakeFiles/Makefile2:140: recipe for target 'client/CMakeFiles/remoteSupport.dir/all' failed
make[1]: *** [client/CMakeFiles/remoteSupport.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2``

guruofquality, 2018-02-23:
Its definitely just a linker error. I guess its not finding the proper libSoapySDR.so. I would build with `make VERBOSE=1` to see that linker arguments are being passed. That way, hopefully you can tell what wrong.

I dont have a ton of experience cross compiling (mostly using qemu myself), but I would be interested in fixing the cmake stuff so its easier to cross compile modules like this.

unixpunk, 2018-02-23:
Thanks, gives me something to start with!  This is the same SoapySDR tree I used to build SoapyPlutoSDR as well, but it has been some time, I'll go back and see what I can find with the verbose option and will also provide you all the variables I have to set in cmake to even get this far.  :)  Side note, I've found it much easier to cross-compile sources that use autoconf instead of cmake.

unixpunk, 2018-02-24:
Interesting...I recompiled SoapySDR and SoapyPlutoSDR which then gave me success in compiling SoapyRemote this time.  Good news!  If you're interested in the cmake information, I've put some txt files here: https://github.com/unixpunk/PlutoSDR/tree/master/SoapyRemote

Let me know if you want any more info.  The overlay folder is all the 'make install' files that will get overlayed on the plutosdr rootfs image.  I'll keep you posted on my results.

unixpunk, 2018-02-24:
It works!  I'm able to get 3msps usable with analog audio, a bit choppy but certainly usable.  Image should be up by tomorrow.

unixpunk, 2018-02-24:
Image working.  In case anyone watching has a PlutoSDR, an OTG Ethernet adapter and is interested: https://github.com/unixpunk/PlutoSDR/tree/master/SoapyRemote