pothosware/SoapyRemote #38: Access SoapyRemote from the public ip ?

santiaarana, 2018-02-25:
Thank you very much for your help with the installation of soapy remote. I have been listening to GQrx and win7 pro from my local network for 19 days and I am very satisfied. Will it be possible soon to access SoapyRemote from the public ip ?. Is there any other sdr client that works from windows 7?

guruofquality, 2018-11-11:
Thats probably why this is an open issue, the control is TCP on a configurable server port. But the streaming (which can be TCP or UDP) just uses the first port available which is allocated by the OS.

You can change this by changing the bind port to something specific that you can put into your NAT/router.

* https://github.com/pothosware/SoapyRemote/blob/master/server/ClientHandler.cpp#L339
* https://github.com/pothosware/SoapyRemote/blob/master/client/Streaming.cpp#L246

Keep in mind that both sides are binding to a socket, and this isnt a case of [UDP hole punching](https://en.wikipedia.org/wiki/UDP_hole_punching). So if its an rx stream, the server is running a server for the rx stream, but the client is also running a server for the flow control messages, and status messages. You can skip status probably, but both sides need a way to get to the port on either side through NAT.

npaisnel, 2018-11-12:
Ok, thanks
Looks a bit more than I’d have any idea how to do.

I’ll stick with the rtlsdr stick and software for now.  Bought an SDRPLAY RSP device as they promised so much more than the rtl usb sticks, but as yet, no software capable of streaming across internet for the RSP devices.   At moment I can tune and stream to iPhone via RTL -TCP stream, which is my ultimate goal

RTL-SDR Receiver Iphone app.

unixpunk, 2019-01-06:
Thanks for the work on this!  Maybe the ship has sailed, but ideally the current protocol would be changed to be "one-way".  Meaning the client (sdr#/cubicsdr) would be the only one making the connection to the server running the SDR.  Maybe you'd also have more flexibility to allow manually entered SoapyRemote addresses rather than only using avahi?  Happy to help brainstorm and test if you'd like.  My interest here would be the ability to place plutosdr devices running SoapyRemote directly on them (like PlutoWeb) facing the web, similar to how Airspy has done.  With a OTG ethernet adapter PlutoSDR can even be headless, no computer needed on the SDR side.  (cpu usage would be of concern as well in my use-case due to the limited nature of embedded SoC's.)

Would also have to take into consideration the bandwidth required of the IQ data obviously, would be very easy to change a setting and have 10MB/sec start coming down the wire.  :)

unixpunk, 2019-01-06:
Just an example, but don't really see why it couldn't: http://www.andrej-mohar.com/plugin-basics-for-sdr

ast, 2019-01-06:
Thanks for the link. Someone should definitely work on a SDR# plugin. It would increase the audience a lot... Unfortunately I don't know C# nor have Windows.

guruofquality, 2019-01-07:
If I had to choose between starting a new project or expanding soapyremote, I would expand it since I think all of the hard stuff of unix sock cross platformness and wrapping every API call has been taken care of.

Mentioned above, its probably an afternoons work to get the NAT issue solved, its a matter of ports and who first binds the sockets and who first sends to that socket. But with that in mind, its a unencrypted protocol with no security or permissions whatsoever :-) 

Here is a list of what I think is missing in SoapyRemote to make it internet safe:

**Security**
* It needs ssl and credentials for the connection and control
* Permissions based on credentials. So once you can connect to the server, which device can you use? which streams can be accessed, which settings can be used, getters only, etc..

**Locating servers**
We wouldn't use mdns/ssdp obviously. But other than explicitly specifying the server, it would be cool to provide a static list of servers as like a conf file on the PC. Some kind of server list that could be sync'd or downloaded from the web. That way apps like cubicsdr could enumerate all of the devices out there on the "cloud" or whatever.

**Changes to streaming**
* we probably want the ability to share the same rx stream to multiple clients
* as mentioned earlier in the issue, the socket setup is a little different because of NATing

ajorg, 2020-09-16:
Hello, I ran into the same issue and I'm wondering if UPnP could be used so that SoapyRemote can open the ports on the firewall. Either of [MiniUPnP](https://github.com/miniupnp/miniupnp) or [GUPnP](https://gitlab.gnome.org/GNOME/gupnp/) could be used to talk to the firewall when establishing connections that appear to be from a remote network.

I would not *require* security be added to the server before supporting this, though obviously that's always a good idea.

kevinelliott, 2021-01-23:
Any chance there is a list of public SoapyRemote SDRs? If not, I would be willing to build and maintain a directory for them, similar to Spyserver Network.