pothosware/SoapyRemote #40: Does SoapyRemote work with soapy_power?

jlobingier1, 2018-04-09:
I am able to find and probe SoapyRemote from a client, but when I install soapy_power on that client it will not connect and says that SoapySDR does not exist....which it does,.  How have others been able to use soapy_power with a SoapyRemote server?

guruofquality, 2018-04-09:
What if you plug in the sdr directly to the client machine, are you saying that is working?

Whats the actual error message, can you share it?

Whats installed on the client and server? `SoapySDRUtil --info`

What options were the softwares configured with? cmake output?

jlobingier1, 2018-04-09:
Josh, thank you for responding. Yes, if I plug a dongle into the client
machine you see what is in the 09-55-59.png attachment below.  Also there
is the response when I unplug the dongle.  No error message, it just cant
find the dongle in the remote server.  The SoapySDRUtil --info responses
are in the other attachment, for both the local (bottom screen) and the
remote server (top screen).  On the surface, to me, it appears that the
SoapySDR side of things is installed OK, and that the problem is with the
soapy_power install.  How should the soapy_power and it's dependencies be
installed?  With "sudo pip3 soapy_power" or "python3 setup.py install" on
each soapy_power package?  When the soapy_power was installed with pip3
there were no obvious error messages.  All of this is installed on Ubuntu
16.04 for the client, and RPi3 for the server.  Both machines have
SoapySDR, SoapyRemote and SoapyRTLSDR installed, as well as their
dependencies.

Thank you,

Jack

jlobingier1, 2018-04-09:
> I dont think github will show any of the attachments from email. So no
> idea whats there.
>
> soapy_power has no idea that its using a remote device, it should be
> transparent.
>
> Make sure there is only one copy of libSoapySDR.so installed on the
> system. Many of these bugs are the result of several different versions
> installed in the same search path, with some tools using one version, and
> some using another. And only one of those versions might have the remote
> support installed for example. Happens all of the time.
>
> In regards to installing soapy_power, its just a python module, so any way
> that it gets installed and python can import it should be good.

Josh, sorry to be so thick on this, but I have three: libSoapySDR.so.0.7.0
(twice)    libSoapySDR.so.o.7  in my /usr/local/lib/     directory.  Do i
rm -r them out of the whole system then reinstall DoapySDR ?

Thanks,

Jack

guruofquality, 2018-04-09:
Multiple library files usually come together for development/soversion/libversion like so
```
ls /usr/local/lib/libSoapySDR.so* -1
/usr/local/lib/libSoapySDR.so
/usr/local/lib/libSoapySDR.so.0.7
/usr/local/lib/libSoapySDR.so.0.7.0
```

I was suggesting that maybe some where hiding in /usr/lib instead, maybe a version 0.6 could be present too, maybe the soapy power has its own libraries copied around, not even sure. -- I dont have a lot of visibility here, so the install manifests from the different projects would sure tell a lot about what was installed. Remember, I havent seen any of the email attachments, so I can only barely guess at the problem.

The only other thing I would check is what the python environment is seeing. So if SoapySDRUtil --info looks good, can you print some python calls in the same environment that you import soapy_power and SoapySDR with? Like print (SoapySDR.getAPIVersion()) and SoapySDR.listSearchPaths(), listModules(), etc. We need to know what version of the libs and modules soapy_power modules and friends are looking it, if at all different.

* Reference https://github.com/pothosware/SoapySDR/blob/master/include/SoapySDR/Modules.hpp
* https://github.com/pothosware/SoapySDR/blob/master/include/SoapySDR/Version.hpp

jlobingier1, 2018-04-09:
>    SoapySDR/Modules.hpp
>    SoapySDR/Version.hpp

Josh, the Soapy install manifests show nothing other than .so, .so.0.7, and
.so.0.7.0.  Here is what --info has:

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.0-g824d83e8
API Version: v0.7.0
ABI Version: v0.7
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/local/lib/SoapySDR/modules0.7/libremoteSupport.so
(0.5.0-3f9666f)
Module found: /usr/local/lib/SoapySDR/modules0.7/librtlsdrSupport.so
(0.2.5-b77a9fc)
Available factories... remote, rtlsdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
jack@jack-RA:~$

jlobingier1, 2018-04-09:
Josh, here is what i get:

>>> import SoapySDR
>>> print(SoapySDR.getAPIVersion())
0.6.0
>>> print (SoapySDR.listSearchPaths())
('/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6',
'/usr/local/lib/x86_64-linux-gnu/SoapySDR/modules0.6',
'/usr/local/lib/SoapySDR/modules0.6'

Where is 0.6 coming from?

jlobingier1, 2018-04-09:
Josh, as a counter to what I just sent you, the same machine shows this
when i look at the directories:

jack@jack-RA:/usr/local/lib/SoapySDR$ ls
modules0.7
jack@jack-RA:/usr/local/lib/SoapySDR$ ls -a
.  ..  modules0.7

guruofquality, 2018-04-09:
You installed two copies of SoapySDR, one from apt package manager with version 0.6, and one from source build (development 0.7). They are both in default search paths, so different tools are picking up different versions. What was the impetus for installing two copies?

My suggestion: 

* use apt repo for library and some modules, build modules from source that are not in the apt repo
        (this also means installing libsoapysdr-dev to get the matching development files)
 * or use all source builds for the main library and modules and remove anything from apt repo

* Similar issue: https://github.com/pothosware/SoapyRemote/issues/31
* Another one: https://github.com/pothosware/SoapySDR/issues/155

jlobingier1, 2018-04-09:
>    Another one: pothosware/SoapySDR#155

Josh, I thought that i installed everything from source.  How do I remove
the items from apt repo?  Thanks for helping me on this.

Jack

jlobingier1, 2018-04-09:
Josh, I guess what i really meant to say was that i thought that the only
things that i didn't install from source were the dependencies that were
listed in the build documents.  How do i determine the packages that need
to be removed?

Thanks again,

Jack

jlobingier1, 2018-04-10:
Josh, running print(SoapySDR.listModules()) I get the following:

>>> import SoapySDR
>>> print(SoapySDR.listModules())
('/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libHackRFSupport.so',
'/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libLMS7Support.so',
'/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libairspySupport.so',
'/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/librtlsdrSupport.so')
>>>

and as far as SearchPaths are concerned, I get the following:

>>> print(SoapySDR.listSearchPaths())
('/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6',
'/usr/local/lib/x86_64-linux-gnu/SoapySDR/modules0.6',
'/usr/local/lib/SoapySDR/modules0.6')
>>>

I know that you sent the suggestions above, but I still don't understand
how to get rid of the 0.6 version if i can't find it.  I have tried to
remove libsoapysdr-dev, but it doesn't exist in my system.  I had GQRX, but
i purged it from the system.  In short, I don't know where this 0.6 version
came from, or how to get rid of it.  Sorry to be so much of a novice on
this, but I'm stuck.

Thank you,

Jack

jlobingier1, 2018-04-11:
Josh, on a clean system, should I be able to install SoapySDR from source, then run the print(SoapySDR.getAPIVersion()) and get a result ?

guruofquality, 2018-04-12:
> Josh, on a clean system, should I be able to install SoapySDR from source, then run the print(SoapySDR.getAPIVersion()) and get a result ?

If its compiled with python bindings, and the PYTHONPATH is set properly, yet.

> I know that you sent the suggestions above, but I still don't understand how to get rid of the 0.6 version if i can't find it. I have tried to remove libsoapysdr-dev, but it doesn't exist in my [...]

If you arent sure about all this stuff, building from source is probably the last thing to do, unless there is a good reason. There are debian packages for soapy remote as well, given everything already installed, it was just another command/checkbox away:

https://github.com/pothosware/PothosCore/wiki/Ubuntu#sdr-runtime-packages

I seriously suggest spending sometime understanding ubuntu/debian packages tools, and some software development patterns on linux, and how the software packages you are using relate to one another. Things will click a lot easier.

jlobingier1, 2018-04-12:
Josh, thanks for your help.  I gave up trying to install from source and just loaded the full Pothosware environment for Ubuntu.  We can close this issue.

https://github.com/pothosware/PothosCore/wiki/Ubuntu#sdr-runtime-packages

closed 2018-04-25 by guruofquality