pothosware/SoapyRemote #52: Will not compile under FreeBSD 11.2

b4, 2018-10-28:
It mainly complains about -lthr not being present in the linker invocation.

-- The CXX compiler identification is Clang 6.0.0
-- The C compiler identification is Clang 6.0.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Looking for include file winsock2.h
-- Looking for include file winsock2.h - not found
-- Looking for include file ws2tcpip.h
-- Looking for include file ws2tcpip.h - not found
-- Looking for include file netdb.h
-- Looking for include file netdb.h - found
-- Looking for include file unistd.h
-- Looking for include file unistd.h - found
-- Looking for include file netinet/in.h
-- Looking for include file netinet/in.h - found
-- Looking for include file netinet/tcp.h
-- Looking for include file netinet/tcp.h - found
-- Looking for include file sys/types.h
-- Looking for include file sys/types.h - found
-- Looking for include file sys/socket.h
-- Looking for include file sys/socket.h - found
-- Looking for include file arpa/inet.h
-- Looking for include file arpa/inet.h - found
-- Looking for include file ifaddrs.h
-- Looking for include file ifaddrs.h - found
-- Looking for include file net/if.h
-- Looking for include file net/if.h - found
-- Looking for include file fcntl.h
-- Looking for include file fcntl.h - found
-- Found Avahi: /usr/local/lib/libavahi-common.so
-- AVAHI_INCLUDE_DIRS=/usr/local/include
-- AVAHI_LIBRARIES=/usr/local/lib/libavahi-common.so;/usr/local/lib/libavahi-client.so
-- Module remoteSupport configured with version: 0.5.0
-- SoapyRemote version: v0.5.0-unknown
-- Install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mar/sdr/SoapyRemote-soapy-remote-0.5.0/build
[mar@impure ~/sdr/SoapyRemote-soapy-remote-0.5.0/build]$ make
Scanning dependencies of target SoapySDRRemoteCommon
[  3%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyURLUtils.cpp.o
[  7%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyRPCSocket.cpp.o
[ 11%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyRPCPacker.cpp.o
[ 14%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyRPCUnpacker.cpp.o
[ 18%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyStreamEndpoint.cpp.o
[ 22%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyHTTPUtils.cpp.o
[ 25%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapySSDPEndpoint.cpp.o
[ 29%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyIfAddrs.cpp.o
[ 33%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyIfAddrsUnix.cpp.o
[ 37%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyInfoUtils.cpp.o
[ 40%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyMDNSEndpointAvahi.cpp.o
[ 44%] Linking CXX static library libSoapySDRRemoteCommon.a
[ 44%] Built target SoapySDRRemoteCommon
Scanning dependencies of target remoteSupport
[ 48%] Building CXX object client/CMakeFiles/remoteSupport.dir/Registration.cpp.o
[ 51%] Building CXX object client/CMakeFiles/remoteSupport.dir/Settings.cpp.o
[ 55%] Building CXX object client/CMakeFiles/remoteSupport.dir/Streaming.cpp.o
[ 59%] Building CXX object client/CMakeFiles/remoteSupport.dir/LogAcceptor.cpp.o
[ 62%] Building CXX object client/CMakeFiles/remoteSupport.dir/ClientStreamData.cpp.o
[ 66%] Building CXX object client/CMakeFiles/remoteSupport.dir/DiscoverServers.cpp.o
[ 70%] Building CXX object client/CMakeFiles/remoteSupport.dir/Version.cpp.o
[ 74%] Linking CXX shared module libremoteSupport.so
[ 74%] Built target remoteSupport
Scanning dependencies of target SoapySDRServer
[ 77%] Building CXX object server/CMakeFiles/SoapySDRServer.dir/ThreadPrioUnix.cpp.o
[ 81%] Building CXX object server/CMakeFiles/SoapySDRServer.dir/SoapyServer.cpp.o
[ 85%] Building CXX object server/CMakeFiles/SoapySDRServer.dir/ServerListener.cpp.o
[ 88%] Building CXX object server/CMakeFiles/SoapySDRServer.dir/ClientHandler.cpp.o
[ 92%] Building CXX object server/CMakeFiles/SoapySDRServer.dir/LogForwarding.cpp.o
[ 96%] Building CXX object server/CMakeFiles/SoapySDRServer.dir/ServerStreamData.cpp.o
[100%] Linking CXX executable SoapySDRServer
/usr/bin/ld: undefined reference to symbol `pthread_create@@FBSD_1.0' (try adding -lthr)
//lib/libthr.so.3: could not read symbols: Bad value
c++: error: linker command failed with exit code 1 (use -v to see invocation)

This is aside from this error which I hit in the past five(!) tagged versions downloaded, across five different compilers. How the hell does this build on linux?!

Scanning dependencies of target SoapySDRRemoteCommon
[  3%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyURLUtils.cpp.o
[  7%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyRPCSocket.cpp.o
/home/mar/sdr/SoapyRemote-soapy-remote-0.5.0/common/SoapyRPCSocket.cpp:503:12: error: no viable conversion from returned value of type 'int' to function return type 'std::string'
      (aka 'basic_string<char, char_traits<char>, allocator<char> >')
    return strerror_r(err, buff, sizeof(buff));
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/v1/string:767:5: note: candidate constructor not viable: no known conversion from 'int' to 'const std::__1::basic_string<char> &' for 1st argument
    basic_string(const basic_string& __str);
    ^
/usr/include/c++/v1/string:772:5: note: candidate constructor not viable: no known conversion from 'int' to 'std::__1::basic_string<char> &&' for 1st argument
    basic_string(basic_string&& __str)
    ^
/usr/include/c++/v1/string:782:31: note: candidate constructor not viable: no known conversion from 'int' to 'const char *' for 1st argument
    _LIBCPP_INLINE_VISIBILITY basic_string(const _CharT* __s);
                              ^
/usr/include/c++/v1/string:815:5: note: candidate constructor not viable: no known conversion from 'int' to 'initializer_list<char>' for 1st argument
    basic_string(initializer_list<_CharT> __il);
    ^
1 error generated.

b4, 2018-10-28:
VERBOSE=1 shows:
/usr/bin/c++  -O3 -DNDEBUG   CMakeFiles/SoapySDRServer.dir/ThreadPrioUnix.cpp.o CMakeFiles/SoapySDRServer.dir/SoapyServer.cpp.o CMakeFiles/SoapySDRServer.dir/ServerListener.cpp.o CMakeFiles/SoapySDRServer.dir/ClientHandler.cpp.o CMakeFiles/SoapySDRServer.dir/LogForwarding.cpp.o CMakeFiles/SoapySDRServer.dir/ServerStreamData.cpp.o  -o SoapySDRServer -Wl,-rpath,/usr/local/lib: /usr/local/lib/libSoapySDR.so -lrt ../common/libSoapySDRRemoteCommon.a /usr/local/lib/libSoapySDR.so /usr/local/lib/libavahi-common.so /usr/local/lib/libavahi-client.so
/usr/bin/ld: undefined reference to symbol `pthread_create@@FBSD_1.0' (try adding -lthr)
//lib/libthr.so.3: could not read symbols: Bad value

Linking manually does happily complete. So just needs -lthr added, and the C++ error properly patched.

guruofquality, 2018-10-28:
> Linking manually does happily complete. So just needs -lthr added, and the C++ error properly patched.

OK so issue one is pthread linking:

So freebsd needs an ifdef here for lthr just like `CMAKE_COMPILER_IS_GNUCXX`
https://github.com/pothosware/SoapyRemote/blob/master/CMakeLists.txt#L50

> no viable conversion from returned value of type 'int' to function return type 'std::string'
> return strerror_r(err, buff, sizeof(buff));

Another issue is the API differences for stderror_r

It needs an addition ifdef for freebsd here (same as OSX): 
https://github.com/pothosware/SoapyRemote/blob/master/common/SoapyRPCSocket.cpp#L499

guruofquality, 2018-10-28:
@b4 can you try this branch https://github.com/pothosware/SoapyRemote/tree/freebsd_fixes

I can manually specify the -lthr flag, but I thought the cmake stuff for finding thread libs would be worth a shot: https://cmake.org/cmake/help/v3.1/module/FindThreads.html

b4, 2018-10-28:
That branch fixed both issues.

closed 2019-01-05 by guruofquality