pothosware/SoapyRemote #54: Custom port when starting from systemd

SDRplay, 2018-11-02:
@guruofquality can you suggest a way to customise the port when starting the server via systemd?

@npaisnel is looking for this to autostart on RPi - the Odroid headless images I've produced all use the default port.

Thanks,

Andy.

npaisnel, 2018-11-02:
Guessing I’ll eventually get an Odroid, they look damn good
But for now, just get the Pi working

Or even tell me what the default port is for SoapyRemote on the Pi

guruofquality, 2018-11-02:
Edit the file and pass an argument to --bind https://github.com/pothosware/SoapyRemote/blob/master/system/SoapySDRServer.service.in#L7 example `--bind="0.0.0.0:1234"` or `--bind="[::0]:1234"`

Thats the quick answer, going forward if this is a useful thing to do, it should be possible to provide a default conf file and include it with EnvironmentFile. So it just comes down to editing a conf file.

guruofquality, 2018-11-02:
> Or even tell me what the default port is for SoapyRemote on the Pi

When a port isnt specified in the server or client, it assumes 55132:

```
//! The default bind port for the remote server
#define SOAPY_REMOTE_DEFAULT_SERVICE "55132"
```

SDRplay, 2018-11-02:
I think a conf file is the way to go if possible. Thanks for the answer, this should help @npaisnel

closed 2018-11-02 by SDRplay

npaisnel, 2018-11-11:
55132 brilliant

if there is a default port I have no need to change it.

I did ask this question on the forum but no one knew.

Thank you