pothosware/SoapyRemote #57: Builing on Alpine Linux 3.8

nmaster2042, 2018-12-29:
I get an issue while building SoapyRemote on Alpine Linux 3.8 X86_64.

alpine:~/SoapyRemote/build# cmake ..
-- The CXX compiler identification is GNU 8.2.0
-- The C compiler identification is GNU 8.2.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Performing Test HAS_STD_CXX11
-- Performing Test HAS_STD_CXX11 - Success
-- Looking for include file winsock2.h
-- Looking for include file winsock2.h - not found
-- Looking for include file ws2tcpip.h
-- Looking for include file ws2tcpip.h - not found
-- Looking for include file netdb.h
-- Looking for include file netdb.h - found
-- Looking for include file unistd.h
-- Looking for include file unistd.h - found
-- Looking for include file netinet/in.h
-- Looking for include file netinet/in.h - found
-- Looking for include file netinet/tcp.h
-- Looking for include file netinet/tcp.h - found
-- Looking for include file sys/types.h
-- Looking for include file sys/types.h - found
-- Looking for include file sys/socket.h
-- Looking for include file sys/socket.h - found
-- Looking for include file arpa/inet.h
-- Looking for include file arpa/inet.h - found
-- Looking for include file ifaddrs.h
-- Looking for include file ifaddrs.h - found
-- Looking for include file net/if.h
-- Looking for include file net/if.h - found
-- Looking for include file fcntl.h
-- Looking for include file fcntl.h - found
-- Found Avahi: /usr/lib/libavahi-common.so
-- AVAHI_INCLUDE_DIRS=/usr/include
-- AVAHI_LIBRARIES=/usr/lib/libavahi-common.so;/usr/lib/libavahi-client.so
-- Found Git: /usr/bin/git (found version "2.20.1")
-- Module remoteSupport configured with version: 0.5.1-5a647db
-- SoapyRemote version: v0.5.1-g5a647db9
-- Install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /root/SoapyRemote/build

Then

alpine:~/SoapyRemote/build# make
Scanning dependencies of target SoapySDRRemoteCommon
[  3%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyURLUtils.cpp.o
[  7%] Building CXX object common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyRPCSocket.cpp.o
/root/SoapyRemote/common/SoapyRPCSocket.cpp: In function 'std::__cxx11::string errToString(int)':
/root/SoapyRemote/common/SoapyRPCSocket.cpp:503:22: error: could not convert 'strerror_r(((int)err), ((char*)(& buff)), sizeof (buff))' from 'int' to 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'}
     return strerror_r(err, buff, sizeof(buff));
            ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [common/CMakeFiles/SoapySDRRemoteCommon.dir/build.make:76: common/CMakeFiles/SoapySDRRemoteCommon.dir/SoapyRPCSocket.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:91: common/CMakeFiles/SoapySDRRemoteCommon.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

I don't know what is the problem.

I used to build on Ubuntu without errors. 

Do you have an idea to fix this ?

guruofquality, 2018-12-29:
You have the variant that returns in int, but the ifdef didnt correctly deduce that: https://github.com/pothosware/SoapyRemote/blob/master/common/SoapyRPCSocket.cpp#L497

* https://linux.die.net/man/3/strerror_r

I'm testing a new check because I dont think the ifdefs are the best check here, can you try this branch: https://github.com/pothosware/SoapyRemote/tree/strerror_r

nmaster2042, 2018-12-29:
Just tried, it passes now.

closed 2018-12-30 by guruofquality