pothosware/SoapyRemote #64: Frequency correction of remote SDR

jujun, 2019-02-08:
Hello,
Thank you for this great software.
It does not seem possible to set a frequency correction in PPM from the client and the driver arguments. It's also not possible to do it on the server side.
Maybe it should be implemented somewhere ?

Thank you
J

guruofquality, 2019-02-08:
No its implemented client and server:

* https://github.com/pothosware/SoapyRemote/blob/master/client/Settings.cpp#L386
* https://github.com/pothosware/SoapyRemote/blob/master/server/ClientHandler.cpp#L661

You will have to share more details, otherwise thats all I can really say

jujun, 2019-02-08:
So maybe the problem is the lack of documentation about this feature. How should it be set ?
If I get it working I can write the documentation.

guruofquality, 2019-02-08:
So I tested that the call at least runs through SoapyRemote `python3 -c "import SoapySDR; SoapySDR.Device().setFrequencyCorrection(SoapySDR.SOAPY_SDR_RX, 0, 0)"` But I dont have my RTLSDR here to check that device specifically which does implement the API call.

The call is documented, at least as far as doxygen is concerned: http://pothosware.github.io/SoapySDR/doxygen/latest/classSoapySDR_1_1Device.html#a3027c581dd2c7527ca92c7630efccad8

you also asked this on IRC and I responded:
```
jujun APP [10:20 AM]
 Hello :)

I search everywhere how to add a frequency correction in ppm for a soapy remote SDR and I cannot find a way to do it, maybe it's not the correct way to do it ?
jujun APP [10:27 AM]
 Maybe I should do it on the server, but I dosent seem possible
josh_blum [10:29 AM]
 The api call setFrequencyCorrection() I think already exists, so you may want to make sure that your device implements this call
```

So perhaps the code for your device does not implement the call setFrequencyCorrection(). If this is from a repo on github, please open an issue for that project, or even better, implement the feature, test it and submit a pull request.

Here is an example implementation in RTLSDR: https://github.com/pothosware/SoapyRTLSDR/blob/master/Settings.cpp#L223

Is there a reason you think SoapyRemote is the problem? Does setFrequencyCorrection() work when the device is used locally, but not through the remote network layer? Otherwise lets close this if SoapyRemote is working fine.

jujun, 2019-02-08:
Thank you for your detailed answer. And sorry I didn't saw your answer on IRC. I need to dig more in the software to understand where the information is not passed, and it seems that the problem is totally out of the scope of the soapy remote project.

guruofquality, 2019-02-08:
Cool, well if you find something wrong with SoapyRemote, just reopen the issue.

If something is wrong with one of the other public github projects, make sure to say something. What commands you ran, what the error message was, what was the hardware. Thats the only way things can get fixed, if someone knows about it :-)

closed 2019-02-08 by guruofquality