pothosware/SoapyRemote #71: hi there

kbtang88, 2019-05-16:
i'm getting this

Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:0.0.0.0]:63433)
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:0.0.0.0]:63664)
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:0.0.0.0]:63665)
SoapyServerListener::close()

i'm not getting to work on cubicsdr

guruofquality, 2019-05-16:
> SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv(header) FAIL: recv() [104: Connection reset by peer]

What is happening on the client side? This could be the client segfaulting, and hence, the server gets abruptly disconnected.  A stack trace of the crash might help, for exampling running cubicsdr in `gdb`.

kbtang88, 2019-05-16:
Soapyremote is on clan and my MacBook is on another vlan. It's trying to
connecting it but cubricsdr sometime crash.

iw5ejm, 2019-05-29:
same thing here, running soapyserver on raspbian (ubuntu 18 based), cubicsdr on macos 10.14.4

guruofquality, 2019-05-29:
Just so everyone knows, the client or server hanging up on a connection is a super generic error which probably means the connection was lost or the application crashed for any number of reasons.

To actually fix something, one would need a backtrace of the crash, or a log of debug prints, something pointing to the last function that was called, what its parameters were. This could be a bug in SoapyRemote client, or for that matter in cubicsdr itself.