pothosware/SoapyRemote #72: SoapyRemote only works on localhost

BeJay8675309, 2019-05-25:
I have an issue where I can't seem to get it working over the network I have a hackrf and a Pi2 as the server and several linux desktops that can't see it at all. SoapyServer runs fine on localhost though. For example on the server I can start the server:
`pi@sdr:~ $ sudo SoapySDRServer --bind 192.168.10.10
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.5.2-gd11da72a
Server UUID: 44948858-c55b-15a1-8567-01cc007f0101
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server... 
Connecting to DNS-SD daemon... 
[WARNING] SoapyRemote compiled without DNS-SD support!
Press Ctrl+C to stop the server
SoapyServerListener::accept([::ffff:192.168.10.10]:34646)
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.10]:34650)
SoapyServerListener::close()
`
and can access the SoapyServer from the server pi machine only (ie it's own IP address - 192.168.10.10):
`pi@sdr:~ $ SoapySDRUtil --find="remote=tcp://localhost:55132, driver=remote"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = HackRF One
  driver = remote
  label = HackRF One #0 a06063c8240a4314
  part_id = a000cb3c007f4f62
  remote = tcp://localhost:55132
  remote:driver = hackrf
  serial = 0000000000000000a06063c82b1e5313
  version = 2015.07.2
`

Trying from any other machine results in nothing:
`me@linux-desktop2:~$ SoapySDRUtil --find="remote=tcp://192.168.10.10:1234, driver=remote"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

No devices found!
me@linux-desktop2:~$ 
`
I have rebuilt the pi from scratch, and tested the network with a usb RTL2832U rtl-tcp server and gqrx which works fine, but I want to get my hackrf to work, so I 'm trying Soapy server. Any ideas why I can't see this from any clients on the same LAN?

Thanks

Bj

guruofquality, 2019-05-25:
Whats SoapySDRUtil --info say on the client machines?

BeJay8675309, 2019-05-25:
Thanks for the reply, here is the output:

`me@linux-desktop2:~$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-gce848c99
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8 (missing)
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
me@linux-desktop2:~$ 
`

BeJay8675309, 2019-05-25:
and on the server it's:

`pi@sdr:~ $ sudo SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-gce848c99
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libHackRFSupport.so (0.3.3-3c514ce)
Module found: /usr/local/lib/SoapySDR/modules0.8/libremoteSupport.so (0.5.2-d11da72)
Available factories... hackrf, remote
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
pi@sdr:~ $ 
`

Do I need to install the hackrf driver on the client? I was hoping it just ports though to gqrx like the rtl server did, or I am missing something?

BeJay8675309, 2019-05-25:
Here is another client with the Soapyhackrf installed, but still can't connect to the server:

`me@linux-desktop1 ~ $ SoapySDRUtil --find="remote=tcp://192.168.10.10:55132, driver=remote"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

No devices found!
me@linux-desktop1 ~ $ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-gce848c99
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libHackRFSupport.so (0.3.3-3c514ce)
Available factories... hackrf
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
`

BeJay8675309, 2019-05-25:
Got it to work! 

I needed to install **SoapyRemote** on the **client** as well.

Now i have:

Available factories... hackrf, remote

Done!

closed 2019-05-25 by BeJay8675309

cr1tbit, 2021-02-20:
Hello, 

In my case, I also had to install soapyRemote package on my client (notebook with manjaro), and only then SoapySDRUtil started recognizing remote devices. Before, it just returned "no devices". The lack of any verbose error message made this whole thing mildly confusing.

I'd like to edit an wiki page to explicitly mention that soapyRemote is required on both server and client but before I do, I'd like to ensure that's not just an edge case, specific for the package in arch repos.

Wiki page:
https://github.com/pothosware/SoapyRemote/wiki

guruofquality, 2021-02-21:
> I'd like to edit an wiki page to explicitly mention that soapyRemote is required on both server and client but before I do, I'd like to ensure that's not just an edge case, specific for the package in arch repos.

Maybe some of the implementation details give it away, but yes, the package needs to be installed on both ends

https://github.com/pothosware/SoapyRemote/wiki#implementation-details