pothosware/SoapyRemote #74: Remote RTL-SDR Direct Sampling?

c5e3, 2019-11-15:
i want to use an rtl-sdr with direct sampling enabled remotely, but i can't get it working with the following device strings:
`soapy=0,driver=remote,remote:driver=rtlsdr,remote=192.168.1.x,direct_samp=2`
`soapy=0,driver=remote,remote:driver=rtlsdr,remote=192.168.1.x,remote:direct_samp=2`

is this feature not implemented or am i doing something wrong?

guruofquality, 2019-11-15:
This mode is set via the `writeSetting()` API call, not the device constructor args

c5e3, 2019-11-15:
thanks for the quick reply!
so there's no chance of using direct sampling with soapyremote + gqrx as a client? or am i missing something?

guruofquality, 2019-11-15:
You can call writeSetting() or any API call from python or C/C++ while using your device in GQRX. Thats what soapyremote does, many clients can connect to the same device. Alternatively, a PR for SoapyRTLSDR to support the setting as a device args string would be cool too :-)

jazzkutya, 2019-11-15:
you can try this https://github.com/jazzkutya/soapysettingsgui please report any problems, I welcome feedback over there

c5e3, 2019-11-15:
i already wrote 2 scripts for my needs, thanks to both of you! I'll definitely try out your gui @jazzkutya 
```import SoapySDR
sdr = SoapySDR.Device("driver=remote,remote:driver=rtlsdr,remote=192.168.1.17")
sdr.writeSetting("direct_samp", "2")
sdr.writeSetting("digital_agc", "true")```
and
```import SoapySDR
sdr = SoapySDR.Device("driver=remote,remote:driver=rtlsdr,remote=192.168.1.17")
sdr.writeSetting("direct_samp", "0")
sdr.writeSetting("digital_agc", "false")```

closed 2019-11-15 by c5e3