pothosware/SoapyRemote #75: Crash of Cubic/Ubuntu and not able to use from SdrAngel/Win10?

9a4db, 2019-12-26:
Please advise?
SoapySDRUtil reporting all god but not able to
reach unit from Win10/SDRAngel or Ubuntu 16.04/CubicSDR
Regards
Djani

// Terminal copy from RPI side
pi@raspberrypi:~ $ SoapySDRServer --bind
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.5.2-g6d9bd820
Server UUID: bc79164c-028a-15e4-8567-7c40007f0101
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server... 
Connecting to DNS-SD daemon... 
[INFO] Avahi version:  avahi 0.7
[INFO] Avahi hostname: raspberrypi-2
[INFO] Avahi domain:   local
[INFO] Avahi FQDN:     raspberrypi-2.local
[INFO] avahi_entry_group_add_service(raspberrypi-2._soapy._tcp)
Press Ctrl+C to stop the server
// Win10 SDRAngel request, API 192.168.10.106: 55132 
SoapyServerListener::accept([::ffff:192.168.10.103]:64711)
SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv() FAIL: header word
SoapyServerListener::close()

// Testing from Ubuntu, Cubic SDR, crash 
SoapyServerListener::accept([::ffff:192.168.10.197]:37350)
SoapyServerListener::accept([::ffff:192.168.10.197]:37352)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.197]:37356)
SoapyServerListener::accept([::ffff:192.168.10.197]:37358)
SoapyServerListener::accept([::ffff:192.168.10.197]:37360)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.197]:37362)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.197]:37368)
SoapyServerListener::accept([::ffff:192.168.10.197]:37370)
SoapyServerListener::accept([::ffff:192.168.10.197]:37372)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.197]:37374)
SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv(header) FAIL: 

//  After that testing with SoapySDRUtil from Ubuntu, terminal copy of RPI side
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.197]:37410)
SoapyServerListener::accept([::ffff:192.168.10.197]:37412)
SoapyServerListener::close()
SoapyServerListener::close()

// Ubuntu side
/Dev/CubicSDR/build/x64$ SoapySDRUtil --find="192.168.10.106=raspberrypi-2, driver=remote"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.002.000-3-g122bfae1

Found device 0
  addr = 24607:1027
  driver = remote
  label = LimeNET-Micro [USB 2.0] 58399D024CE9
  media = USB 2.0
  module = FT601
  name = LimeNET-Micro
  remote = tcp://192.168.10.106:55132
  remote:driver = lime
  serial = 0058399D024CE9
q@q-X756UX:~/Dev/CubicSDR/build/x64$

9a4db, 2019-12-26:
[image]
Cubic can see remote unit, but crashes on selection...

guruofquality, 2019-12-26:
The failure message on the server may just be the result of the client application dying. Do you have a backtrace for the client application crash, that may help point in the right direction?

Also what does SoapySDRUtil --info say on the client side?

9a4db, 2019-12-26:
Thank you for prompt response :+1: 
// pi side
pi@raspberrypi:~ $ SoapySDRServer --bind
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.5.2-g6d9bd820
Server UUID: 6d6e1957-0a9c-15e4-8567-0ade007f0101
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server... 
Connecting to DNS-SD daemon... 
[INFO] Avahi version:  avahi 0.7
[INFO] Avahi hostname: raspberrypi-2
[INFO] Avahi domain:   local
[INFO] Avahi FQDN:     raspberrypi-2.local
[INFO] avahi_entry_group_add_service(raspberrypi-2._soapy._tcp)
Press Ctrl+C to stop the server
SoapyServerListener::accept([::ffff:192.168.10.197]:38940)
SoapyServerListener::accept([::ffff:192.168.10.197]:38942)
SoapyServerListener::close()
SoapyServerListener::close()

// Ubuntu side
q@q-X756UX:~/Dev/CubicSDR/build/x64$ SoapySDRUtil --find="192.168.10.106=raspberrypi-2, driver=remote"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.002.000-3-g122bfae1

Found device 0
  addr = 24607:1027
  driver = remote
  label = LimeNET-Micro [USB 2.0] 58399D024CE9
  media = USB 2.0
  module = FT601
  name = LimeNET-Micro
  remote = tcp://192.168.10.106:55132
  remote:driver = lime
  serial = 0058399D024CE9

q@q-X756UX:~/Dev/CubicSDR/build/x64$

9a4db, 2019-12-26:
// Ubuntu side - Cubic SDR
q@q-X756UX:~/Dev/CubicSDR/build/x64$ ./CubicSDR
Loading:: configuration file '/home/q/.CubicSDR/config.xml'

Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

SDR enumerator starting.
SoapySDR init..
	API Version: v0.6.0
	ABI Version: v0.6
	Install root: /usr/local
	Loading modules... 
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.002.000-3-g122bfae1

	Available factories...lime, null, remote, sdrplay, uhd
Available vertical sync SwapInterval functions: 
	glxSwapIntervalEXT: No
	DRI2SwapInterval: No
	glxSwapIntervalMESA: No
	glxSwapIntervalSGI: No
No vertical sync swap interval functions available.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono12_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono16_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono18_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono24_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono27_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono32_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/local/share/cubicsdr/fonts/vera_sans_mono36_0.png', parsed 255 characters.
  addr = 24607:1027
  driver = remote
  label = LimeNET-Micro [USB 2.0] 58399D024CE9
  media = USB 2.0
  module = FT601
  name = LimeNET-Micro
  remote = tcp://192.168.10.106:55132
  remote:driver = lime
  serial = 0058399D024CE9
Make device 0
[INFO] Make connection: 'LimeNET-Micro [USB 2.0] 58399D024CE9'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeNET-Micro
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M register cache: Disabled
  boardSerialNumber=0x9d024ce9
  firmwareVersion=6
  gatewareVersion=1.0
  hardwareVersion=19
  protocolVersion=1
  remote:version=0.5.2-g6d9bd820

Reporting enumeration complete.
SDR enumerator done.
[INFO] Make connection: 'LimeNET-Micro [USB 2.0] 58399D024CE9'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeNET-Micro
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M register cache: Disabled
Terminating SDR thread..
ERROR: thread '9SDRThread' has not terminated in time ! (> 3000 ms)
q@q-X756UX:~/Dev/CubicSDR/build/x64$ 

// RPI side
SoapyServerListener::accept([::ffff:192.168.10.197]:39058)
SoapyServerListener::accept([::ffff:192.168.10.197]:39060)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.197]:39064)
SoapyServerListener::accept([::ffff:192.168.10.197]:39066)
SoapyServerListener::accept([::ffff:192.168.10.197]:39068)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.197]:39070)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.197]:39126)
SoapyServerListener::accept([::ffff:192.168.10.197]:39128)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.197]:39134)
SoapyServerListener::accept([::ffff:192.168.10.197]:39136)
SoapyServerListener::accept([::ffff:192.168.10.197]:39138)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.197]:39140)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.197]:39148)
SoapyServerListener::accept([::ffff:192.168.10.197]:39150)
SoapyServerListener::accept([::ffff:192.168.10.197]:39152)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.197]:39154)
SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv(header) FAIL: 
SoapyServerListener::close()
SoapyServerListener::close()

guruofquality, 2019-12-26:
Can you share the output of `ls /usr/lib/x86_64-linux-gnu/libSoapy* /usr/local/lib/libSoapy* /usr/include/Soapy* /usr/local/include/Soapy*`

I didnt expect to see 0.6, its a little old. I figure you would have compiled this soapysdr and friends from source recently. I'm just wondering if this is an ABI issue, you may just need to clean and recompile the software.

9a4db, 2019-12-26:
Here it is:
q@q-X756UX:~/Dev/CubicSDR/build/x64$ ls /usr/lib/x86_64-linux-gnu/libSoapy* /usr/local/lib/libSoapy* /usr/include/Soapy* /usr/local/include/Soapy*
/usr/lib/x86_64-linux-gnu/libSoapyOsmoSDR.so.0
/usr/lib/x86_64-linux-gnu/libSoapyOsmoSDR.so.0.2.4
/usr/lib/x86_64-linux-gnu/libSoapySDR.so
/usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.6
/usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.6.1
/usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7
/usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7.1
/usr/local/lib/libSoapySDR.so
/usr/local/lib/libSoapySDR.so.0.6
/usr/local/lib/libSoapySDR.so.0.6.1

/usr/include/SoapySDR:
Config.h                 Device.h    Formats.hpp  Registry.hpp  Version.h
Config.hpp               Device.hpp  Logger.h     Time.h        Version.hpp
Constants.h              Errors.h    Logger.hpp   Time.hpp
ConverterPrimitives.hpp  Errors.hpp  Modules.h    Types.h
ConverterRegistry.hpp    Formats.h   Modules.hpp  Types.hpp

/usr/local/include/SoapySDR:
Config.h     Device.hpp  Formats.hpp  Modules.hpp   Types.h
Config.hpp   Errors.h    Logger.h     Registry.hpp  Types.hpp
Constants.h  Errors.hpp  Logger.hpp   Time.h        Version.h
Device.h     Formats.h   Modules.h    Time.hpp      Version.hpp
q@q-X756UX:~/Dev/CubicSDR/build/x64$

9a4db, 2019-12-26:
Just to notice that I have the same problem on Win10 machine running SDRAngel.
SDRAngel 4.11.10 is not crashing, just remote is not working
// RPI side
pi@raspberrypi:~ $ SoapySDRServer --bind
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.5.2-g6d9bd820
Server UUID: 8db7482b-0c6f-15e4-8567-0dd9007f0101
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server... 
Connecting to DNS-SD daemon... 
[INFO] Avahi version:  avahi 0.7
[INFO] Avahi hostname: raspberrypi-2
[INFO] Avahi domain:   local
[INFO] Avahi FQDN:     raspberrypi-2.local
[INFO] avahi_entry_group_add_service(raspberrypi-2._soapy._tcp)
Press Ctrl+C to stop the server
SoapyServerListener::accept([::ffff:192.168.10.103]:53391)
SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv() FAIL: header word
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.10.103]:53392)
SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv() FAIL: header word
SoapyServerListener::close()

guruofquality, 2019-12-26:
theres at least 3 versions of SoapySDR on that one machine. I recommend reading this: https://github.com/pothosware/SoapySDR/wiki/ConfigGuide#avoid-simultaneous-installs

9a4db, 2019-12-27:
OK, thank you, let's go from scratch, starting with clean Ubuntu 18.04 LTS.
Will report the outcome.

9a4db, 2019-12-28:
All is up and running. New 18.04 installation is able to see remote devices.
To be on safe on Pi side, used Sdrplay image, striped down and Lime package installed. 
Now able to run SDRPlay RSP1 or Lime Net on  Soapy remote at last on CubicSDR

9a4db, 2019-12-29:
Now Ubuntu 18.04 story, clean installation:

q@q-X756UX:/opt/build/sdrangel/build$ ls /usr/lib/x86_64-linux-gnu/libSoapy* /usr/local/lib/libSoapy* /usr/include/Soapy* /usr/local/include/Soapy*
/usr/lib/x86_64-linux-gnu/libSoapySDR.so  /usr/local/lib/libSoapySDR.so.0.8
/usr/local/lib/libSoapySDR.so             /usr/local/lib/libSoapySDR.so.0.8.0

/usr/include/SoapySDR:
Config.h                 Device.h    Formats.hpp  Registry.hpp  Version.h
Config.hpp               Device.hpp  Logger.h     Time.h        Version.hpp
Constants.h              Errors.h    Logger.hpp   Time.hpp
ConverterPrimitives.hpp  Errors.hpp  Modules.h    Types.h
ConverterRegistry.hpp    Formats.h   Modules.hpp  Types.hpp

/usr/local/include/SoapySDR:
Config.h                 Device.h    Formats.hpp  Registry.hpp  Version.h
Config.hpp               Device.hpp  Logger.h     Time.h        Version.hpp
Constants.h              Errors.h    Logger.hpp   Time.hpp
ConverterPrimitives.hpp  Errors.hpp  Modules.h    Types.h
ConverterRegistry.hpp    Formats.h   Modules.hpp  Types.hpp
q@q-X756UX:/opt/build/sdrangel/build$ 

q@q-X756UX:~$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################
Found device 0
  addr = 24607:1027
  driver = remote
  label = LimeNET-Micro [USB 2.0] 58399D024CE9
  media = USB 2.0
  module = FT601
  name = LimeNET-Micro
  remote = tcp://192.168.10.106:55132
  remote:driver = lime
  serial = 0058399D024CE9
Found device 1
  driver = remote
  label = SDRplay Dev0 RSP1 B0001P0002
  remote = tcp://192.168.10.106:55132
  remote:driver = sdrplay
  serial = B0001P0002
q@q-X756UX:~$ 

q@q-X756UX:~$ /opt/install/sdrangel/bin/sdrangel
2019-12-29 15:51:56.615 (I) SDRangel 4.12.4 Qt 5.9.5 64b x86_64 Ubuntu 18.04.3 LTS DSP Rx:24b Tx:16b PID 25067
2019-12-29 15:51:56.615 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/q/.config/f4exb/SDRangel.conf
2019-12-29 15:51:56.615 (D) AudioDeviceManager::AudioDeviceManager: scan input devices
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #0: default
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #1: pulse
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #2: sysdefault:CARD=PCH
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #3: front:CARD=PCH,DEV=0
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #4: dmix:CARD=PCH,DEV=0
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #5: dsnoop:CARD=PCH,DEV=0
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #6: hw:CARD=PCH,DEV=0
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #7: plughw:CARD=PCH,DEV=0
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #8: alsa_input.pci-0000_00_1f.3.analog-stereo
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: input device #9: alsa_output.pci-0000_00_1f.3.analog-stereo.monitor
2019-12-29 15:51:56.762 (D) AudioDeviceManager::AudioDeviceManager: scan output devices
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #0: default
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #1: pulse
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #2: sysdefault:CARD=PCH
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #3: front:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #4: surround21:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #5: surround40:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #6: surround41:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #7: surround50:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #8: surround51:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #9: surround71:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #10: hdmi:CARD=PCH,DEV=0
2019-12-29 15:51:57.377 (D) AudioDeviceManager::AudioDeviceManager: output device #11: hdmi:CARD=PCH,DEV=1
[... 242 lines cut ...]
2019-12-29 15:52:00.041 (I) WebAPIServer::start: starting web API server at http://127.0.0.1:8091
2019-12-29 15:52:00.041 (D) MainWindow::MainWindow: end
2019-12-29 15:52:00.060 (D) GLSpectrum::initializeGL: context:  major:  3  minor:  0  ES:  no
2019-12-29 15:52:29.392 (D) MainWindow::sampleSourceChanged: tab at 0
2019-12-29 15:52:29.393 (D) DeviceAPI::saveSamplingDeviceSettings: not a suitable preset
2019-12-29 15:52:29.393 (D) DSPDeviceSourceEngine::stopAcquistion
2019-12-29 15:52:29.393 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStop
2019-12-29 15:52:29.393 (D) DSPDeviceSourceEngine::gotoIdle
2019-12-29 15:52:29.396 (D) FileInput::stop
2019-12-29 15:52:29.403 (D) SoapySDRInput::openDevice: allocated SampleFifo
2019-12-29 15:52:29.404 (D) SoapySDRInput::openDevice: open device here
2019-12-29 15:52:29.404 (D) DeviceSoapySDR::openopenSoapySDRFromSequence: driver=remote
[... cut ...]
[INFO] Make connection: 'LimeNET-Micro [USB 2.0] 58399D024CE9'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeNET-Micro
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M calibration values caching Disable
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
Aborted (core dumped)
q@q-X756UX:~$ 

Crash on selection of any of present remote devices, my guess is that trouble may
be on SDRAngel instalation side, therefore same issue opened on SDRAngel git...

9a4db, 2019-12-30:
With kind assistance of author finally managed to run CubicSDR over the Soapy Remote

[image]

On the other hand SDRangel have some bug in relation of Soapy Remote use, units are visible, but
on selection SDRangel crash. So far author is not ready to look in that problem, here his answer:
"
Then use Cubic SDR if it works fine for you. You do not  need Soapy for LimeSDR that is natively supported. Soapy suppport is a  workaround for devices that are not natively supported in case this may  be useful. I will not make any effort concening Soapy.
"
[image]