pothosware/SoapyRemote #76: Crash on failure to create avahi client or poller

dicta, 2020-03-19:
In at least some failure cases of avahi, avahi_client_new() can return a null pointer, even when we set the NO_FAIL flag on this call. This causes immediate crashes of SoapySDR because `SoapyMDNSEndpoint` assumes the client exists. The following stack trace was observed:

```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[New Thread 0x7ffff4b30700 (LWP 128340)]
[Thread 0x7ffff4b30700 (LWP 128340) exited]
[New Thread 0x7ffff432f700 (LWP 128341)]
[New Thread 0x7ffff3b2e700 (LWP 128342)]
[ERROR] avahi_client_new() failed: Daemon not running
SoapySDRUtil: client.c:785: avahi_client_get_state: Assertion `client' failed.
[New Thread 0x7ffff4b30700 (LWP 128343)]
[New Thread 0x7ffff2b2c700 (LWP 128345)]
[New Thread 0x7ffff332d700 (LWP 128344)]

Thread 3 "SoapySDRUtil" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffff432f700 (LWP 128341)]
0x00007ffff7bdace5 in raise () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff7bdace5 in raise () from /usr/lib/libc.so.6
#1  0x00007ffff7bc4857 in abort () from /usr/lib/libc.so.6
#2  0x00007ffff7bc4727 in __assert_fail_base.cold () from /usr/lib/libc.so.6
#3  0x00007ffff7bd3426 in __assert_fail () from /usr/lib/libc.so.6
#4  0x00007ffff79bcdca in avahi_client_get_state () from /usr/lib/libavahi-client.so.3
#5  0x00007ffff7a2fdb1 in SoapyMDNSEndpoint::status() () from /usr/lib/SoapySDR/modules0.7/libremoteSupport.so
#6  0x00007ffff7a16e8f in ?? () from /usr/lib/SoapySDR/modules0.7/libremoteSupport.so
#7  0x00007ffff79faf01 in ?? () from /usr/lib/SoapySDR/modules0.7/libremoteSupport.so
```

The code block that can fail is the following (common/SoapyMDNSEndpointAvahi.cpp around line 80). Note also that failure of the avahi_simple_poll_new() call will also trigger a crash since client will never be created in this failure case.

```
SoapyMDNSEndpointData::SoapyMDNSEndpointData(void):
    simplePoll(nullptr),
    client(nullptr),
    group(nullptr),
    browser(nullptr),
    resolversInFlight(0),
    browseComplete(false)
{
    simplePoll = avahi_simple_poll_new();
    if (simplePoll == nullptr)
    {
        SoapySDR::logf(SOAPY_SDR_ERROR, "avahi_simple_poll_new() failed");
        return;
    }

    int error(0);
    client = avahi_client_new(avahi_simple_poll_get(simplePoll), AVAHI_CLIENT_NO_FAIL, &clientCallback, this, &error);
    if (client == nullptr or error != 0)
    {
        SoapySDR::logf(SOAPY_SDR_ERROR, "avahi_client_new() failed: %s", avahi_strerror(error));
        return;
    }
}
```

guruofquality, 2020-03-22:
I cant replicate client being null on my system, but I think I can handle working around it pretty easily, does this branch work for you? https://github.com/pothosware/SoapyRemote/tree/avahi_client_null

MountainLogic, 2020-07-06:
This is a note in a bottle for anyone using Ubuntu on the windows linux subsystem with SoapySDR and not a request for a bug fix.  This is not directly related to SoapyRemote.

I am able to repo this bug (#76) under the windows linux subsystem with Ubuntu:
Distributor ID: Ubuntu                                                                                                                                                                                               Description:    Ubuntu 18.04.4 LTS                                                                                                                                                                                   Release:        18.04                                                                                                                                                                                                Codename:       bionic   
Radio:            E4000

I followed the install instructions https://github.com/pothosware/PothosCore/wiki/Ubuntu and then built lime driver following the directions in Eric's answer here: https://stackoverflow.com/questions/55850748/how-to-properly-install-soapysdr-with-limesdr-usb

This frankenstein windows/linux environment is probably not worth untwisting and I will likely move back to a real linux environment.  I also ran rx_power & rx_SDR that was compiled with (I think) GCC.  (https://github.com/rxseger/rx_tools) and got weird results with UHD reporting Win32.  I had little expectations that this would ever run the signal chain all the way out to real hardware , but wow, did it find the windows driver?  On the WIndows host I've installed a number of the usual GnuRadio tools, so perhap it really did find the windows driver - which would be amazing.  This would be the first time that WIN32 "exceeded expectations" ;-) 

```
foo@LAP-60922:~/wip/rx_tools/build$ rx_power.exe -f 88M:108M:125k -
Number of frequency hops: 8
Dongle bandwidth: 2500000Hz
Downsampling by: 1x
Cropping by: 0.00%
Total FFT bins: 256
Logged FFT bins: 256
FFT bin size: 78125.00Hz
Buffer size: 16384 bytes (3.28ms)
Reporting every 10 seconds
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.15.0.0-0-gaea0e2de
Found Elonics E4000 tuner
Using device Audio: device_id=0 origin=https://github.com/pothosware/SoapyAudio
Found 1 channel(s) :
Channel 0 :
  Found 1 antenna(s): RX
  Found 0 gain(s):
  Found 1 frequencies: RF
  Found 14 sample rates: 4000 5512 8000 9600 11025 16000 22050 32000 44100 48000 88200 96000 176400 192000
  Found 0 bandwidths:
[INFO] Using format CS16.
```
which did return reasonable data.

rx_sdr, however, did fail:

```
foo@LAP-60922:~/wip/rx_tools/build$ rx_sdr.exe -f 1900M -
dup2 start: No error
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.15.0.0-0-gaea0e2de
Found Elonics E4000 tuner
Using device Audio: device_id=0 origin=https://github.com/pothosware/SoapyAudio
Found 1 channel(s) :
Channel 0 :
  Found 1 antenna(s): RX
  Found 0 gain(s):
  Found 1 frequencies: RF
  Found 14 sample rates: 4000 5512 8000 9600 11025 16000 22050 32000 44100 48000 88200 96000 176400 192000
  Found 0 bandwidths:
Using output format: CU8 (input format CS16, 4 bytes per element)
Sampling at 2048000 S/s.
Tuned to 1900000000 Hz.
[INFO] Using format CS16.
Reading samples in sync mode...
Failed to activate stream
```

But SoapySDRUtil fails.  I've not investigated the cause.
```
foo@LAP-60922:~/wip/rx_tools/build$ SoapySDRUtil
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Usage SoapySDRUtil [options]
  Options summary:
    --help                               Print this help message
    --info                               Print module information
    --find[="driver=foo,type=bar"]       Discover available devices
    --make[="driver=foo,type=bar"]       Create a device instance
    --probe[="driver=foo,type=bar"]      Print detailed information

  Advanced options:
    --check[=driverName]                 Check if driver is present
    --sparse                             Simplified output for --find

  Rate testing options:
    --args[="driver=foo"]                Arguments for testing
    --rate[=stream rate Sps]             Rate in samples per second
    --channels[="0, 1, 2"]               List of channels, default 0
    --direction[=RX or TX]               Specify the channel direction

foo@LAP-60922:~/wip/rx_tools/build$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[ERROR] avahi_client_new() failed: Daemon not running
SoapySDRUtil: client.c:785: avahi_client_get_state: Assertion `client' failed.
Aborted (core dumped)
```

guruofquality, 2020-07-06:
This change should have been pushed earlier, sorry about that 

So avahi failing to make a client should not kill the discovery

* https://github.com/pothosware/SoapyRemote/commit/b15941d4e188eb76f749ab373eb20ff98e6d3655

closed 2020-07-06 by guruofquality