pothosware/SoapyRemote #77: Crash after connect

jorisrobijn, 2020-03-21:
Hi,

I have a SDRplay duo, and am running SoapySDRremote on a raspberry pi. When I connect with CubicSDR, from a windows PC on the same network, it crashes. I can connect to the device when I connect it locally on the PC.

BTW, why are there ao many new threads being spawned and exited all the time?

```
(gdb) run --bind
Starting program: /usr/local/bin/SoapySDRServer --bind
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.5.2-g6d9bd820
Server UUID: e7462cf5-5691-15fe-8567-4072007f0101
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server...
[New Thread 0x76b7c040 (LWP 16504)]
Connecting to DNS-SD daemon...
[WARNING] SoapyRemote compiled without DNS-SD support!
Press Ctrl+C to stop the server
SoapyServerListener::accept([::ffff:10.18.20.81]:58431)
[New Thread 0x761ff040 (LWP 16523)]
[Thread 0x761ff040 (LWP 16523) exited]
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:10.18.20.81]:58433)
[New Thread 0x761ff040 (LWP 16575)]
[New Thread 0x73c4c040 (LWP 16576)]
[New Thread 0x7344b040 (LWP 16577)]
[Thread 0x73c4c040 (LWP 16576) exited]
[Thread 0x7344b040 (LWP 16577) exited]
[New Thread 0x72aff040 (LWP 16578)]
[New Thread 0x7344b040 (LWP 16579)]
[Thread 0x7344b040 (LWP 16579) exited]
[Thread 0x72aff040 (LWP 16578) exited]
[Thread 0x761ff040 (LWP 16575) exited]
SoapyServerListener::accept([::ffff:10.18.20.81]:58434)
[New Thread 0x72aff040 (LWP 16581)]
SoapyServerListener::close()
[New Thread 0x761ff040 (LWP 16582)]
[Thread 0x761ff040 (LWP 16582) exited]
[Thread 0x72aff040 (LWP 16581) exited]
SoapyServerListener::accept([::ffff:10.18.20.81]:58435)
[New Thread 0x761ff040 (LWP 16583)]
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:10.18.20.81]:58436)
[New Thread 0x72aff040 (LWP 16584)]
[New Thread 0x7344b040 (LWP 16585)]
[Thread 0x7344b040 (LWP 16585) exited]
[New Thread 0x7344b040 (LWP 16586)]
[Thread 0x7344b040 (LWP 16586) exited]
[New Thread 0x7344b040 (LWP 16587)]
[New Thread 0x73c4c040 (LWP 16588)]
[Thread 0x73c4c040 (LWP 16588) exited]
[Thread 0x7344b040 (LWP 16587) exited]
[Thread 0x761ff040 (LWP 16583) exited]
[Thread 0x72aff040 (LWP 16584) exited]
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:10.18.20.81]:58439)
[New Thread 0x72aff040 (LWP 16607)]
[New Thread 0x761ff040 (LWP 16608)]
[Thread 0x761ff040 (LWP 16608) exited]
[Thread 0x72aff040 (LWP 16607) exited]
SoapyServerListener::accept([::ffff:10.18.20.81]:58440)
[New Thread 0x761ff040 (LWP 16610)]
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:10.18.20.81]:58441)
[New Thread 0x72aff040 (LWP 16611)]
[New Thread 0x7344b040 (LWP 16612)]
[Thread 0x7344b040 (LWP 16612) exited]
[New Thread 0x7344b040 (LWP 16613)]
[Thread 0x7344b040 (LWP 16613) exited]
[New Thread 0x7344b040 (LWP 16614)]
[New Thread 0x720ff040 (LWP 16615)]
[New Thread 0x7187c040 (LWP 16622)]
[New Thread 0x7107b040 (LWP 16623)]
[Thread 0x7107b040 (LWP 16623) exited]
[New Thread 0x704ff040 (LWP 16624)]

Thread 25 "SoapySDRServer" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7187c040 (LWP 16622)]
0x76fba418 in memcpy () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so
(gdb) bt
#0  0x76fba418 in memcpy () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so
#1  0x3fe00000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

guruofquality, 2020-03-22:
I cant tell much from the backtrace, if you run this locally on the raspi when the server is running do you get the crash?
* `SoapySDRUtil --find="driver=remote,remote=localhost"`
*  `SoapySDRUtil --make="driver=remote,remote=localhost"`

Whats the output of `SoapySDRUtil --info` on the raspi

Any duplication install stuff that would cause ABI issues on the raspi, like the same packages from source and from deb packages while compiling? Like libsoapysdr installed from a deb, and actually development files for soapysdr from source.

> BTW, why are there ao many new threads being spawned and exited all the time?

Its parallelizing the discovery stuff with std::future(), which is spawning the reads

jorisrobijn, 2020-03-23:
I don't think there are packages also present compiled from source. The Soapy and SDRplay stuff were installed from git source. 
The --find and --make commands don't crash the server.

Below is output from SoapySDRUtil --info
```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-g980c2c50
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libremoteSupport.so  (0.5.2-6d9bd82)
Module found: /usr/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so (0.3.0-14ec39e)
Available factories... remote, sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

```

guruofquality, 2020-03-28:
> The --find and --make commands don't crash the server.

Thats good to hear, this could easily be cubic calling a setting on the SoapySDRplay thats crashing rather than a general problem. There could be a call set segfaults on arm or something like that.

Running the server in gdb is the right idea. I think if you build SoapyRemote and SoapySDRplay with debug and get a backtrace, it should show the call. And maybe its something we can report to the respective issue tracker.

crazyclimber999, 2022-11-22:
I have a similar issue:

root@raspberrypi:/home/pi# SoapySDRServer --bind
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.5.2-unknown
Server UUID: 0249ece2-0333-172a-8567-059a007f0101
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server... 
Connecting to DNS-SD daemon... 
[INFO] Avahi version:  avahi 0.8
[INFO] Avahi hostname: openwebrxbrugherio
[INFO] Avahi domain:   local
[INFO] Avahi FQDN:     openwebrxbrugherio.local
[INFO] avahi_entry_group_add_service(openwebrxbrugherio._soapy._tcp)
Press Ctrl+C to stop the server
SoapyServerListener::accept([fe80::10b1:c6c8:5ac8:b0%3]:50966)
SoapyServerListener::accept([::ffff:172.30.2.18]:50967)
[INFO] [UHD] linux; GNU C++ version 10.2.0; Boost_107100; UHD_3.15.0.0-4
usb_claim_interface error -6
Found Rafael Micro R820T tuner
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([fe80::10b1:c6c8:5ac8:b0%3]:50969)
SoapyServerListener::accept([fe80::10b1:c6c8:5ac8:b0%3]:50970)
SoapyServerListener::close()
SoapyServerListener::accept([fe80::10b1:c6c8:5ac8:b0%3]:50971)
SoapyServerListener::accept([fe80::10b1:c6c8:5ac8:b0%3]:50972)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([fe80::10b1:c6c8:5ac8:b0%3]:50973)
SoapyServerListener::close()
SoapyServerListener::accept([fe80::10b1:c6c8:5ac8:b0%3]:50974)
terminate called after throwing an instance of 'char const*'
Aborted
root@raspberrypi:/home/pi# 

this happened when I search the device in CUBICSDR

crazyclimber999, 2024-01-21:
up

zuckschwerdt, 2024-01-21:
There is no (date) sorting in the issues, a bump won't do anything but annoy people.

We can't reproduce and need more info: where does that `throw` of a string (e.g. `throw "the error string";`) originate? It's not SoapyRemote, all throws are `std::runtime_error` here.