pothosware/SoapyRemote #78: Tips for wifi ?

markdrayton247, 2020-04-07:
I'm attempting to stream from soapySdr to my mac running Cubic. Has anyone got any experience of doing this over wifi ? The stream is pretty choppy. Actually it was choppy over ethernet until I switched up the mtu to 4000 but I can't really do this with wifi.... can I throttle the bandwidth requirement somehow ? changing sample rate doesn't seem to help. thanks

mikenye, 2020-12-30:
Anecdotally, I’ve read that people have decreased choppiness by slowing the waterfall right down to its lowest speed. I’m not sure if this is placebo or not. 👍 
I found this issue also looking for possible fixes. :-)

gonesurfing, 2021-01-29:
I have it working OK over wifi:
-Pi3 and Pi3+ on RTL end. Pi3+ has gigabit, and helps a little but it wasn't terrible with 100BASE-Tx
-MacBook Pro on 11.1 (Big Sur)
-Ubiquity AP-AC-lite access points.

Setting MTU to 4000 helps, but probably once every 5min I still get a stutter.