pothosware/SoapyRemote #79: Crash with more than 31 network interfaces

chaplin89, 2020-05-08:
Hi, it seems to me there is a bug in the function [SoapyRPCSocket::selectRecvMultiple](https://github.com/pothosware/SoapyRemote/blob/2a7363e485745153aa80eee4f912d5abf2d4e6cc/common/SoapyRPCSocket.cpp#L480): for what I can see, the function is creating a mask on a 32bit integers using N bits, where N is the number of the sockets that is related to the network interfaces. 

Since the function is using signed 32 bit integers, 31 interfaces is the maximum possible. If that function returns a negative value, it's interpreted as an error by the caller.

Unfortuntely for me I'm using more than 31 interfaces (it'a thing related to cuttlefish, didn't had the time to investigate) and this is making the library crash badly.

Questions:
1) is my analysis correct?
2) are there any workaround I can use?
3) If my analysis is correct, how to fix that? If there was a way to filter out virtual network interfaces I think it would be a great starting point.

guruofquality, 2020-05-09:
Can you try this branch? https://github.com/pothosware/SoapyRemote/tree/increase_select_multiple

chaplin89, 2020-05-18:
Yes, it seems to work.
Thanks 
[image]

[image]

closed 2020-05-18 by guruofquality