pothosware/SoapyRemote #86: CubicSDR can connect but cannot listen.

mkierczak, 2020-11-30:
Got a problem talking to my Airspy from CubicSDR via SoapyRemote, I can connect, but there is no waterfall. I installed from sources on a clean system installation on Ubuntu. Details also [here](https://github.com/cjcliffe/CubicSDR/issues/862). Cheers! Martin.

nickoe, 2021-12-17:
@mkierczak Did you ever get a resolution to this? I am trying the same on two arch machines, but with gqrx instead of cubicsdr. On the remote the device is probed just fine wiht SoapySDRUtil --probe, and on the client (connecting to the temote) I also see it when probing, AND gqrx can find the deice, but no samples show up in the waterfall.  This is when I test on my home LAN. I am not sure what could be the cause of this. Could it be something like the remote not really starting to sample the airspy?

mkierczak, 2022-03-04:
No, sadly I have never resolved it really...

mkierczak, 2022-03-09:
OK, so I have some follow-up. Still no success though, but some issues solved. Following t[his excellent text](https://0xfeed.tech/2017/04/remote-access-tools-for-sdr-receivers/#_solutions_for_sdr_remote_access), I found out that my ssh tunnelling might be an issue and I added `remote:prot=tcp` to the device string. My setup:

1. Establish a tunnel: `ssh user@remote -L 55132:127.0.0.1:55132`2. 
2. On the remote: `SoapySDRServer --bind="tcp://127.0.0.1:55132"`
3. On the local: `rx_fm -f 92.3M -M wbfm -r 32k -g 35 -d driver=remote,remote=tcp://127.0.0.1:55132,remote:driver=airspyhf,remote:timeout=100000,'remote:prot'='tcp' | play -t s16 -c 1 -r 32k -b 16 -`
4. Output on the local:
```
-: (s2)

 File Size: 0
  Encoding: Signed PCM
  Channels: 1 @ 16-bit
Samplerate: 32000Hz
Replaygain: off
  Duration: unknown

In:0.00% 00:00:00.00 [00:00:00.00] Out:0     [      |      ]        Clip:0    Using device AirspyHF: remote:version=0.6.0-gc09b2f10 serial=dd52eb5df83839e2
Found 1 channel(s) :
Channel 0 :
  Found 1 antenna(s): RX
  Found 2 gain(s): LNA RF
  Found 1 frequencies: RF
  Found 4 sample rates: 768000 384000 256000 192000
  Found 0 bandwidths:
[INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CS16, scaleFactor=32768, mtu=4096, window=16384)
[INFO] Server side stream bound to 127.0.0.1:35953
SoapySDRDevice_setupStream failed: SoapyRemote::setupStream(tcp://127.0.0.1:35953) -- connect FAIL: connect(tcp://127.0.0.1:35953) [61: Connection refused]
``` 

Output on the remote:
```
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.6.0-gc09b2f10
Server UUID: 4b211209-b9a1-16da-8567-0bf9007f0101
Launching the server... tcp://127.0.0.1:55132
Server bound to 127.0.0.1:55132
Launching discovery server...
Connecting to DNS-SD daemon...
[INFO] Avahi version:  avahi 0.7
[INFO] Avahi hostname: rg
[INFO] Avahi domain:   local
[INFO] Avahi FQDN:     rg.local
[INFO] avahi_entry_group_add_service(rg._soapy._tcp)
Press Ctrl+C to stop the server
SoapyServerListener::accept(127.0.0.1:37864)
SoapyServerListener::accept(127.0.0.1:37866)
SoapyServerListener::close()
SoapyServerListener::accept(127.0.0.1:37868)
SoapyServerListener::accept(127.0.0.1:37870)
SoapyServerListener::close()
# TONS OF THE SAME WITH DIFFERENT PORTS INC BY 2
# OCCASSIONALLY:
SoapyServerListener::accept(127.0.0.1:37948)
SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv(header) FAIL:
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept(127.0.0.1:37950)
SoapyServerListener::accept(127.0.0.1:37952)
```

On the remote, there is firewall running but I did:
`sudo ufw allow from 127.0.0.1 to 127.0.0.1 proto tcp` 
to be sure local connections on any port are allowed.

zuckschwerdt, 2022-03-09:
So the issue here is rather "Redirecting SoapyRemote through SSH", right?
It might be easier to use a VPN, preferably the lightweight Wireguard – you get all ports TCP and UDP there.

closed 2022-04-28 by mkierczak