pothosware/SoapyRemote #87: Timeout on OpenWebRX

n1zzo, 2021-03-09:
When using SoapyRemote with a RTL-SDR Blog V3, with OpenWebRX as a client,
I get a timeout error every 1-3 hours.

While this happens the SoapyRemote stdout is clean:
```
$ SoapySDRServer --bind
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.5.1-unknown
Server UUID: 01ff786f-d886-1667-8567-1486a8c06e04
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server...
Connecting to DNS-SD daemon...
[INFO] Avahi version:  avahi 0.8
[INFO] Avahi hostname: degu-2
[INFO] Avahi domain:   local
[INFO] Avahi FQDN:     degu-2.local
[INFO] avahi_entry_group_add_service(degu-2._soapy._tcp)
Press Ctrl+C to stop the server
SoapyServerListener::accept([::ffff:192.168.4.200]:58286)
Found Rafael Micro R820T/2 tuner
SoapyServerListener::accept([::ffff:192.168.4.200]:58292)
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.4.200]:58294)
Found Rafael Micro R820T/2 tuner
Allocating 15 zero-copy buffers
```

This kind of fault is considered unrecoverable by the OpenWebRX client, [here is the related issue on their side](https://github.com/jketterl/openwebrx/issues/219).

guruofquality, 2021-03-14:
Help me bisect the problem. Can you throw some prints into SoapyRTLSDR readStream? Lets see if the timeout comes from the hardware driver or the SoapyRemote: https://github.com/pothosware/SoapyRTLSDR/blob/master/Streaming.cpp#L340

If the hardware driver actually starts returning timeous, then so will the calls to SoapyRemote's readStream

n1zzo, 2021-03-30:
Hi guruofquality,
sorry for the late reply, I should have performed the tests you asked.
This is the patch I've created which prints the number of read elements:
```
diff --git a/Streaming.cpp b/Streaming.cpp
index cac56e6..703ce7d 100644
--- a/Streaming.cpp
+++ b/Streaming.cpp
@@ -359,6 +359,7 @@ int SoapyRTLSDR::readStream(
     if (bufferedElems == 0)
     {
         int ret = this->acquireReadBuffer(stream, _currentHandle, (const void **)&_currentBuff, flags, timeNs, timeoutUs);
+       fprintf(stderr, "Read %d elements from SDR\n", ret);
         if (ret < 0) return ret;
         bufferedElems = ret;
     }
 ```
After having compiled, installed, and executed the code,
after waiting openwebrx to hang, these are the latest debug messages:

```
Read 131072 elements from SDR
Read -1 elements from SDR
Read 131072 elements from SDR
Read -1 elements from SDR
Read 131072 elements from SDR
Read -1 elements from SDR
Read 131072 elements from SDR
Read -1 elements from SDR
Read 131072 elements from SDR
Read -1 elements from SDR
Read 131072 elements from SDR
Read -1 elements from SDR
Read 131072 elements from SDR
Read -1 elements from SDR
Read 131072 elements from SDR
Read -1 elements from SDR
Read 131072 elements from SDR
```
So it appears that there were no errors from the SDR device and the timeout is coming from SoapyRemote itself, am I correct?

If you need further testing just let me know.