pothosware/SoapyRemote #96: FR: Option to disable mDNS

Artanicus, 2022-11-14:
### Problem:
- Building a minimal SoapySDRServer for example as a Docker image is needlessly complicated due to the assumption that mDNS is always available.
- Not all scenarios call for mDNS, for example if alternative service discovery methods are used. In my scenario I'm actually relying on Consul service discovery.
```
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.5.2-unknown
Server UUID: a28c6e87-882b-1727-8567-000111ac1e00
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server... 
Connecting to DNS-SD daemon... 
[ERROR] avahi_client_new() failed: Daemon not running
Press Ctrl+C to stop the server
DNS-SD daemon disconnected...
Exiting prematurely...
Shutdown client handler threads
Cleanup complete, exiting
```

### Proposed solution:
- Add an optional flag `--disable-mdns` that skips client creation and just skips straight to binding.
- Alternatively, if client creation fails, still proceed with serving, just without mDNS. This can however lead to inconsistent results if mDNS has a transient issue, so my preference is on requiring an explicit opt-in or opt-out for mDNS, whichever is preferred for expected default behavior.

guruofquality, 2022-11-15:
try this branch: with -DENABLE_AVAHI=OFF passed to cmake https://github.com/pothosware/SoapyRemote/pull/97

Artanicus, 2022-11-18:
That seems to work at least as far as being able to compile and start the server without Avahi, which is great! I was also able to get CubicSDR to connect, so I'd say that's a working solution.

closed 2024-01-21 by zuckschwerdt