pothosware/SoapyRemote #104: `SoapySDRUtil --find="remote=a.b.c.d"` instantly throws `No devices found! ` without trying to send a packet.

ipaqmaster, 2024-02-23:
I'm connected to a remote network over a VPN (Routed to the destination network)

`SoapySDRUtil --find="remote=a.b.c.d"` instantly throws `No devices found! ` without trying to send a packet to the remote.

I can reach the remote Soapy service just fine:

```
ncat -v a.b.c.d 55132
Ncat: Version 7.94 ( https://nmap.org/ncat )
Ncat: Connected to a.b.c.d:55132.
```

But watching with `tcpdump` reveals Soapy didn't attempt to send any packets at all.

zuckschwerdt, 2024-02-23:
The `--find` in SoapySDRUtil uses `SoapySDR::Device::enumerate()` which for SoapyRemote is here:
https://github.com/pothosware/SoapyRemote/blob/master/client/Registration.cpp#L51

Can you increase the verbosity (`SOAPY_SDR_LOG_LEVEL=DEBUG SoapySDRUtil ...`) to see "SoapyClient querying devices for %s" and possibly "SoapyRemote::find() -- connect(%s) FAIL: %s"?

ipaqmaster, 2024-02-26:
Well that's a ginormous relief! Today the same command is working, successfully enumerating the remote which has the SDR plugged in with `SoapySDRServer --bind` running from this laptop running a VPN to the remote network and on a different broadcast domain, when the IP is provided. I'm not sure why it's working today and could only potentially point at a recent update on each machine perhaps quietly fixing something. I will keep an eye open for if this happens again.

Trying to use this remote in GQRX seems to 'pause' every second or so either as some kind of underrun or and intolerance to the latency/throughput of funneling raw RF traffic from the remote over Soapy. This is over a low latency, high throughput fibre link though and selecting the lowest SDR sampling rate (240000) doesn't seem to fix the issue. I will look into this on my end as I know this problem does not happen over a 1gbps LAN of sub-1ms-latency.

closed 2024-02-26 by ipaqmaster